artisan-roaster-scope/artisan #88: Add support for Thermoworks BlueTherm Duo

DBrahms, 2016-02-18:
Following up on a post I made Feb 2 on the Artisan blog (http://artisan-roasterscope.blogspot.com/2013/06/device-selection.html)

I have a bluetooth thermocouple probe (Thermoworks BlueTherm Duo) that I've been using for years in the kitchen, and would like to use it with Artisan instead of having to buy a second, "compatible," device for coffee roasts. This device already works with the Roastworks app, but that's a paid app plus it has no PC-based option. Thermoworks also makes a free logging software that I use for cooks, but it doesn't have any coffee-specific functionality. I have followed up with the Thermoworks people directly and they are open to helping integrate their product with Artisan.

BlueTherm device here: http://thermoworks.com/products/logger/bluetherm_duo.html

Thermoworks logging software here: http://thermoworks.com/products/logger/index-studiotest.php

Email response from Val at Thermoworks:

Here is what my engineer said:
"In theory, the developer (behind Artisan) could write a bit of code to take the temperature results from a BlueTherm Duo and integrate it into this program. At this stage, we don’t have any plans to do this."

So it looks like it could be done but whoever is behind Artisan would have to write more code to make it so. 

If you are able to find out who that is and want to give them my information I'll be happy to pick up the conversation with them. 

Best of luck!

Regards,
Val McQuiston
Product Manager & Technical Sales
741 E Utah Valley Drive
American Fork, UT 84003 
Toll-free: 800.393.6434
Ph: 801-756-7705
Fax: 801-756-8948
val.mcquiston@thermoworks.com

MAKOMO, 2016-02-26:
Message to Val sent.

MAKOMO, 2016-04-15:
Negative answer received

closed 2016-04-15 by MAKOMO