artisan-roaster-scope/artisan #92: HeatSnob USB Thermocouple Support

gobears98, 2016-04-15:
I figured out how to interface to the HeatSnob USB thermocouple, using hid_api.
I have 2 devices working well with Artisan, using and external application interface.

If there is any interest in building in official support for the device, I could provide the source code.
Its actually quite simple code, just a few hundred lines.

BTW, the heatsnob device is inexpensive ($60), compact, well designed and very well built.
You can connect as many as your have USB ports.

They are very pupular in Australia where they originate, but also available in the USA.
Not sure about the rest of the world.

Anyway, I am really excited to try out Artisan as an alternative to the RoastMaster software they ship with their product.

Cheers,
~Bruce

MAKOMO, 2016-04-15:
Please share your code and we will see if it can be integrated cross-plattform

gobears98, 2016-04-15:
`/*******************************************************
 HeatSnob USB HID interface code
 * Bruce W. McGaughy
 * 
 * Based on, and following same license as:
 * 
 * Windows HID simplification

 Alan Ott
 Signal 11 Software

 8/22/2009

 Copyright 2009, All Rights Reserved.
 
 This contents of this file may be used by anyone
 for any reason without any conditions and may be
 used as a starting point for your own applications
 which use HIDAPI.
 * 
********************************************************/

#include <stdio.h>
#include <wchar.h>
#include <string.h>
#include <stdlib.h>
#include "hidapi.h"
#include <math.h>

// Headers needed for sleeping.
#ifdef _WIN32
	#include <windows.h>
#else
	#include <unistd.h>
#endif

// Dump verbose information if user passes "info" string to application.
static bool  VERBOSE_MODE = false;

//////////////////////////////////////////////////////////////////////////
// Given the buffer of bytes from reading USB HID of Maxim 31855,, read the 
// thermocouple and internal chip temperature.
// INPUT:  buf = Buffer of all bytes read from USB HID interface.
//               NOTE: The 32 bit output of the Maxim 31855 is stored in 
//                     bytes 3, 4, 5, and 6, with the 0th bit in the right 
//                     most bit of byte 6, and the 31st bit in the left most
//                     bit of byte 3.
// OUTPUTS: tc = The thermocouple temperature reading in celsius.
//          it = Internal temperature of the Maxim chip.
//////////////////////////////////////////////////////////////////////////

void get_temps( unsigned char* buf, double& tc, double& it ){
    
    // Read the bits from bit 31 down to bit 0. The bit_index indicates
    // which bit we are working on.
    int bit_index = 31;
    
    // Initialize the outputs to zero, and accumulate temperature with each bit
    tc = 0.0;
    it = 0.0;
    
    // The most significant bit of TC temperature is 4096 * 0.25 celsius.
    // Each lower bit cuts the tc_mult in half
    double tc_mult = 4096.0;

    // Most signficant bit of internal tempearture is 1024 * .0625 celsius,
    // Each lower bit cuts the it_mult in half.
    double it_mult = 1024.0;

    // Read the bits from 31...0 by iterating each byte from 3...7 and then
[... 140 lines cut ...]
            if ( p == 0 ){
                printf("%03.2f", tc_temps[p] );
            }else{
                printf(",%03.2f", tc_temps[p]);
            }
        }
        
        printf("\n");
        
        if ( VERBOSE_MODE ){
            for ( int p = 0 ; p < num_paths ; ++p ){
                printf("Device Path:  %s,  thermocouple temp: %03.2f,  internal chip temp: %03.2f\n", paths[p], tc_temps[p], it_temps[p] );
            }
        }

	/* Free static HIDAPI objects. */
	hid_exit();

#ifdef WIN32
	system("pause");
#endif

	return 0;
}
`

gobears98, 2016-04-15:
Above is the file.
It is based on the hidtest.c application in the mac directory of hid_api_0.7.0 distribution.

The makefile is there too. Just copy/paste my code above into hidtest.c (in the mac directory), call make, and it should work fine. Since hid_api is cross-platform, and has the similar directories for windows & linux with minimal differences, it should be straightforward to plug-in. Especially if you have an HID USB framework already built into Artisan. Then its even easier.

gobears98, 2016-04-15:
I added some comments to help understand the data layout of the Maxim chip.
And, added support to catch and report faults like open circuit or short to gnd.

I will insert the code into the next message.

gobears98, 2016-04-15:
`/*******************************************************
 HeatSnob USB HID interface code
 * Bruce W. McGaughy
 * 
 * Based on, and following same license as:
 * 
 * Windows HID simplification

 Alan Ott
 Signal 11 Software

 8/22/2009

 Copyright 2009, All Rights Reserved.
 
 This contents of this file may be used by anyone
 for any reason without any conditions and may be
 used as a starting point for your own applications
 which use HIDAPI.
 * 
********************************************************/

#include <stdio.h>
#include <wchar.h>
#include <string.h>
#include <stdlib.h>
#include "hidapi.h"
#include <math.h>

// Headers needed for sleeping.
#ifdef _WIN32
	#include <windows.h>
#else
	#include <unistd.h>
#endif

// Dump verbose information if user passes "info" string to application.
static bool  VERBOSE_MODE = false;

//////////////////////////////////////////////////////////////////////////
// Maxium 31855: Tempearture measurement chip of Heatsnob:
// https://www.adafruit.com/datasheets/MAX31855.pdf
// 
// The chip outputs 32 bits of data, which are packed in 4 chars which are
// in the output buffer of HID read(), starting at 4th char 
//
// Summary of bits: 31 : Sign bit of Thermocouple temperature, 1 = negative
//                  30 : MDB Thermocouple: 1024c = 4096c * 0.25 c 
//                  ...  Bits 29...19 are from 512c ... 0.5c
//                  18 : LSB Thermocouple:  0.25c = 1 * 0.25c 
//                  17 : Reserved, always 0.
//                  16 : Fault bit, 1=fault
//                  15 : Sign bit of internal chip temperature, 1 = negative
//                  14 : Internal Chip Temp MSB: 64c = 1024c * .0625c
//                  ...  Bits 13...5 are from 32c ...  0.25c
//                  4 :  LSB of internal chip temperature:  0.0625
//                  3 :  Reserved, always 0.
//                  2 :  1 = short to vcc
//                  1 :  1 = short to gnd
//                  0 :  1 = open circuit
//////////////////////////////////////////////////////////////////////////

//////////////////////////////////////////////////////////////////////////
// Given the buffer of bytes from reading USB HID of Maxim 31855,, read the 
// thermocouple and internal chip temperature.
// INPUT:  buf = Buffer of all bytes read from USB HID interface.
//               NOTE: The 32 bit output of the Maxim 31855 is stored in 
//                     bytes 4, 5, 6, and 7 (indexes 3,4,5,6), 
//                     with the 0th bit in the right 
//                     most bit of byte 6, and the 31st bit in the left most
[... 191 lines cut ...]
                printf("%03.2f", tc_temps[p] );
            }else{
                // Print the other temperatures for Artisan if available.
                printf(",%03.2f", tc_temps[p]);
            }
        }
        
        printf("\n");
        
        if ( VERBOSE_MODE ){
            for ( int p = 0 ; p < num_paths ; ++p ){
                printf("Device Path:  %s,  thermocouple temp: %03.2f,  internal chip temp: %03.2f\n", paths[p], tc_temps[p], it_temps[p] );
            }
        }

	/* Free static HIDAPI objects. */
	hid_exit();

#ifdef WIN32
	system("pause");
#endif

	return 0;
}
`

bubbaj3, 2017-02-05:
I just downloaded 0.9.9 and the heatsnob prob doesn't seem to work.  Does anyone know if this has been incorporated into the latest release?  I probably don't have it configured correctly.

MAKOMO, 2017-10-20:
The provided code is not in Python and no push was provided. Artisan does for now not provide support for any USB HID devices. This might never change.

closed 2017-10-20 by MAKOMO