artisan-roaster-scope/artisan #95: Support for 4k monitor

MarcelPr, 2016-06-02:
All the dialog windows (such as editing events or working with designer) are small to clearly show the variable numerical values.

MarcelPr, 2016-06-02:
[image]

MarcelPr, 2016-06-02:
[image]

MAKOMO, 2016-12-19:
Could you try the actual v1.0.0b4 builds? Please test both the regular Python2 Windows as well as the Python3 build (artisan-win-1.0.0b4-py3.zip)

utilitalian, 2017-07-19:
I also have this issue running v1.0.1 on my new Surface Pro 4. Temp displays need a microscope to see, and many of the words and numbers are cut off in the windows.

MAKOMO, 2017-10-20:
A first beta version of the Qt5-based Artisan v1.2 has been released now that supports high-resolution screens better.

closed 2017-10-20 by MAKOMO