artisan-roaster-scope/artisan #97: Artisan works after setup, but not after restarting program

slightlyoffbeat, 2016-07-03:
I've used artisan on a MBP and a Macbook Air. I have the latest Hottop k+. 

I followed the setup guide for hottop + artisan and I'm able to see stats and control the hottop. The problem is if I shut down artisan, I'm never able to get it working again. I've tried multiple orders of turning things on/plugging in. Loading settings each time etc.

Stumped on this one. Is there a step I'm missing? 

http://artisan-roasterscope.blogspot.com/2015/05/hottop-kn-8828b-2k.html

As an example, if I just do steps 2 & 3 (which takes 30 seconds) I'm able to see output from the hottop. If I shut down artisan and open it back up, I can confirm all settings are the same (device/serial/extra) but I can't get it to read anything.

Thoughts?

slightlyoffbeat, 2016-07-03:
One important thing I didn't mention: I can only get it to work again by doing a factory reset, and then going through the setup process again. So basically it only works after a factory reset, but once i shut down the program it won't work again unless I factory reset and do the minimum setup again.

MAKOMO, 2016-12-19:
Could you resolve this. If you  are working on Mac OS X > 10.9, don't install the FTDI driver. The system comes already with one. If you did, uninstall the FTDI driver again.

MAKOMO, 2017-07-16:
No further input on this to make this reproducible, so I close this.

closed 2017-07-16 by MAKOMO