artisan-roaster-scope/artisan #99: Error 193 when using a custom python script for temp input

vlee489, 2016-08-29:
I am getting `[Error 193] %1 is not a valid Win32 application @line 28148` when trying to test Artisan with a simple python script that is 

```python
import time
import sys

ET = 0
BT = 0

loop = 1

print ("200,200")

while loop == 1:
    ET = ET + 1
    BT = BT + 1
    print (str(ET),",",str(BT))
    time.sleep(0.5)
````

lukeinator42, 2017-01-19:
@vlee489 the program should output the temperature each time it is called, so you dont want a loop. For example: 
`#!/usr/bin/env python
print ("237.1,100.4")`

also, the script needs to be an executable. On mac or linux you can type this in the terminal:
`chmod +x test.py`
on windows you will probably have to use something like http://www.py2exe.org/

MAKOMO, 2017-07-16:
Thanks lukeinator for your comment. I'll close this as there is no further input on this.

closed 2017-07-16 by MAKOMO