artisan-roaster-scope/artisan #104: Set Up

NYCroaster, 2016-11-02:
Does anyone know how I can change my temperature reading from 00.0 to 000. my temp is reading 400 degrees at 40.0 which in turn gives me a flat graph.
[image]

MAKOMO, 2016-11-05:
Please the artisan mailing list for configuration questions to the community. This issue mechanism is for reporting implementation bugs.

You can add a symbolic formula like "x/10" via menu Config >> Device; 3rd tab "Symb ET/BT" for either "ET Y(x)" or "BT Y(x)", to divide your original values x by 10 if this is what you are asking for.

closed 2016-11-05 by MAKOMO

NYCroaster, 2016-11-10:
Thanks Marko,

do u know how can redo a hard reset on the preferences. I followed the blog
set up for the proaster and now the readings dont even work and I forgot
what the original settings were.

Yeekai

MAKOMO, 2016-11-10:
My last reply here. Please use the the public mailing list: menu Help >> Factory Reset. Once you have a running configuration you should use Help >> Save Settings to store all app settings in a file for later recovery.

NYCroaster, 2016-11-10:
Sorry thnx

Yeekai Lim