artisan-roaster-scope/artisan #105: Problem with use Artisan & FZ94

bzuc0, 2016-11-09:
Hello everybody!

I have little problem with Artisan software.

I use Coffee-Tech FZ94 roaster and i try pair this machine with Artisan software. I buy this machine in this year, it is the newest machine from coffee-tech. When i install virtual ports, that i pair Artisan with machine, i am connected, but software say this error message:

"Modbus Error: readSingleRegister() Exception response @line 28245"

Ill try make anything, i use manual on https://artisan-roasterscope.blogspot.sk/2015/01/connecting-artisan-to-coffee-tech-fz-94.html and https://artisan-roasterscope.blogspot.sk/2016/08/fz-94-1-bumping-up-modbus.html . 

Artisan is connected, but dont show temperature or anything. And of course, send upwriting message.

Thank you for any help!

MAKOMO, 2016-11-10:
I know that their older build from this year were still running at 2400 baud, while the latest batches are running at 19200 baud (or maybe 9600) just test different serial settings.

bzuc0, 2016-11-10:
Thank you for support. Ill try iz and it works! 

Matej

Odoslané z iPhonu

MAKOMO, 2016-12-17:
Thanks for your confirmation!

closed 2016-12-17 by MAKOMO

lukkubiny, 2017-09-19:
Hi Marco,

same problem here today, roasted 5 months without any problem, now still showing "Modbus Error: readSingleRegister() Exception response @line **29640**" 
I tried also different serial settings, nothing changed.

Appreciate any comments,
Luk

MAKOMO, 2017-09-19:
Luk, this seems beyond my scope as I did not change anything on your setup. That error indicates a general communication issue. Artisan seems not to be able to communicate with the PIDs via the USB2RS485 interface. Either you changed something in your software setup (Artisan configuration, OS update, driver update), your serial driver got corrupted, or it is a hardware defect. Sorry for not being of more help here.

lukkubiny, 2017-09-20:
Hi Marco,

thank you anyway :) Will check all possibilities.

Best,
Luk

rollag16, 2019-12-20:
I have a problem artisan 1.1.0 "Modbus Error: readSingleRegister() Exception response @line 30574"

MAKOMO, 2019-12-20:
This issue is closed. Artisan v1.1.0 is no longer supported. Only the latest version can be  supported, which is currently v2.1.1. Most often those MODBUS Exception responses simply indicate a lack of connectivity to the machine. Maybe you did not install the correct USB driver.