artisan-roaster-scope/artisan #110: Import

4thcrack, 2016-12-16:
Hi, how do i import data from probat pilot shop?
i have the xml files.

MAKOMO, 2016-12-17:
There is no import support for files from the Probat Pilot software yet. I add this on the (long) list of feature requests and keep this issue open for now.

MAKOMO, 2016-12-17:
I just implemented Probat Pilot import, including the burner rate. Could you share one of your xml files to test this?

4thcrack, 2016-12-19:
[NeuesZip.zip](https://github.com/artisan-roaster-scope/artisan/files/661371/NeuesZip.zip)

Hi thanks for the answer.
i send you the file.

MAKOMO, 2016-12-19:
Thanks for this. I discovered that I implemented the import/export only for recipes. Added now support for "histories" (import only as the Pilot does not has an import for histories, right?). Sadly the data in those exports has no decimal place and thus the result is not perfect and generates quite some noise on the RoR (a lot more than just recording directly with Artisan from a Probatone!). It is also obvious that something is wrong with the RoR computation in the Pilot software. Anyhow.

This feature will be released as part of Artisan v1.0. Expect more Probat Probatone support coming at the begin of next year;)

[image]
[image]
[image]

closed 2016-12-19 by MAKOMO