artisan-roaster-scope/artisan #113: Multiple Serial Ports Configuration

Nortorf, 2017-01-16:
When you use Artisan for reading ET/BT and Drum/Fan speed together 2 sets of  Comm Port parameters should be given with a communication port (same parameters for both).  If you use USB/RS485 converter to connection you have to replug the USB/RS485 converter after starting Artisan roasting screen with START button. Because it only reads ET/BT data until replug. 
But I use wireless serial server for communication between Artisan and PIDS - inverters and it is not possible replug. 
Please ask me if further information necessary. 
Best Regards
Anton Holl

MAKOMO, 2017-01-17:
Artisan only supports one MODBUS ring with up to 6 devices/channels. To read from more than one MODBUS device, all devices have to by daisy chained to the first MODBUS device on the same ring, accessed via the same USB-2-RS485 converter.

closed 2017-01-17 by MAKOMO

Nortorf, 2017-01-17:
The Modbus devices on the same ring but problem occurs with PIDs. Artisan either reads PIDs or Modbus devices. Only get data from both after replug of the USB-2-RS485. But I use serial server for wireless communication and comm port is virtual and replug is not possible. 
Thanks

MAKOMO, 2017-01-17:
I still don't understand your setup. What is the difference between "PID" and "MODBUS" device? Don't you access both via MODBUS? What type of PID are you talking about? In case you are using a Fuji PID as main device (first tab of device assignment dialog;  menu Config >> Device), and an extra device of type MODBUS (2nd tab of device assignment dialog; menu Config >> Device). In this case, the Fuji is assumed to be on a separate MODBUS ring (configured via Serial Dialog, 1st tab ET/BT) from the MODBUS device ring (configured via Serial Dialog, 3rd tab "MODBUS". You can force the Fuji to use the same ring as the MODBUS device by ticking "MODBUS port" in the Device Assignment Dialog (1st tab, PID section). Than this one ring has to be connected to the serial port configured via the MODBUS tab.

Nortorf, 2017-01-17:
PIDs: Delta DTA series
Modbus Devices: Delta VFD L series
I give both same parameters and same virtual comm port address. There is 2 screen for parameters to be given , one for Delta DTA PIDs, other for Modbus devices.

Nortorf, 2017-01-17:
I will check with Modbus port box on ET/BT screen.

Nortorf, 2017-01-17:
I just tested with Modbus Port check box at ET/BT configuration screen. Problem still continue. If I give a dummy Comm Port at  Serial Port Configuration to ET/BT it reads Modbus devices, vice versa if I give a dummy Comm port to Serial Port Configuration to Modbus devices it reads Drum and Fan speed futures. Simultaneously both not possible unless replug while Artisan roasting screen working. 
Note: On serial server  wireless communication to do replug not possible.

Nortorf, 2017-01-17:
Correction: 
If I give a dummy Comm Port at Serial Port Configuration to ET/BT it reads Modbus devices, vice versa if I give a dummy Comm port to Serial Port Configuration to Modbus devices it reads Delta DTA PIDs. Actually it reads first PIDs comm port.

MAKOMO, 2017-01-17:
Ok, now I understand what you want to achieve. You want to select the DeltaPID as main device and your other MODBUS devices as extra devices, both on the same MODBUS ring. While this is supported (by a hack) for Fuji PIDs, this is not supported for DeltaPIDs. Those are assumed to be on different MODBUS rings, accessed via two different serial ports (connected via 2 separate USB-2-RS485 converters). This has mostly historic reasons. The support for DeltaPID was donated in an early project phase, but the supported "disappeared". So the DeltaPID support is quite minimal anyhow.

However, there is hope. You could access your Delta DTA PIDs (I assume you have two of them, one for BT with slave ID1 and one for ET with slave ID2) by configuring your main Artisan device as MODBUS (chan 1+2) and your Delta VFD  as extra devices (chan 3+4) with slave ID3 and ID4, respectively. 

[image]
[image]

According to the documentation I found, the PV of the Delta DTA can be read of register 4700H, which translates to 18176. SV register is 4701H (dec 18177), register p is 4708H (18184), register i is 4709H (18185) and register d is 470AH (18186). The unit in all cases is 0.1 (so divider 1/10). The PID can be turned off and on by writing 1 (run) or 0 (stop) into register 4719H (18201). Note that all those register numbers might be wrong by an offset of 1. If the above configuration does not work try with a setup where all register numbers are reduced by 1 or increased by 1. There are some strange MODBUS implementation around that do require such an offset by 1.

In the screen shot below, I assume you want to use the BT PID as controlling PID via the SV slider, therefore I added Slave 1 in the PID box. Put 2 for controlling via the ET PID (also in the ON/OFF commands).

[image]

Hope this helps!

Nortorf, 2017-01-18:
It solves problem accept SV. It tried to add parameters -+1. I think there is no data box to display SV.  If I do sets as you mentioned here 
[image]
But if I change Input 2 register values to 18177 I get SV too instead of ET
[image]
If i do as here 
[image]
It displays ET and BT 2 times
[image]

Thanks

MAKOMO, 2017-01-18:
Ok, now you are close and I almost understand what you are trying to achieve;)

SV as in my definition is just set as an "output", such that if you move the SV slider or if Artisan's software PID via its follow-background mechanism does it, the changed value is send to the connected PID. 

If you want to display the current SV as established in the PID, just add another extra device (channel) to read that in (here channel 5).  See the screenshots below. Note that I am using "MODBUS_34" and "MODBUS_56" here as device type to access the MODBUS channels 3,4,5, and 6 as defined in the MODBUS tab. In your screenshot above you used "MODBUS" as type also for the extra device definition, which always returns channels 1 and 2 (so your ET and BT). That is why your Drum and Fan LCDs did not work.

[image]
[image]
[image]

Hope this helps. Please consider to donate to Artisan to enable its further development, especially if you are in a commercial setting. The donate information here on Github on the main project page.

Marko

Nortorf, 2017-01-21:
Thanks all worked great. I was connecting to the roaster remotely via Teamviewer and could not see the SV slider working and set values to the PID. 

Note: I produce commercial coffee roasters.

MAKOMO, 2017-01-21:
> On 21 Jan 2017, at 01:12, Nortorf <notifications@github.com> wrote:
> 
> Note: I produce commercial coffee roasters.

Then please make a service contract with me.

Best,
Marko