artisan-roaster-scope/artisan #114: Wireless Communication Interuption

Nortorf, 2017-01-21:
I use high quality serial server for communicate wirelessly with Artisan. It works great but while processing  some times interrupts and graphic seems in pieces. Is there any solution if communication stops to stay in same digit at curve? 

[image]

Best Regards

poundy, 2017-02-12:
Those points where your temperature drops to -1 just show that your serial port response to Artisan reading the data was too slow.  I'm not aware of any way to have Artisan "average" items and honestly I'd be surprised if that'd ever be something to consider.  You should figure out why the serial data isn't coming through reliably.  For me I know my device that was sending serial data was also writing to a local SD card periodically, and when that and the Artisan read crossed over I'd lose data.  Fixing that meant no more gaps in my graphs.

MAKOMO, 2017-07-03:
A patch was commit that improves on this. Drop outs are still recorded as -1, but don't influence the graph any longer. However, your communication seems to be heavily disturbed and you should fix this on the hardware side or try a slower sampling interval.

closed 2017-07-03 by MAKOMO