artisan-roaster-scope/artisan #115: Using logging to file inside of callprogram() python scripts

int3ll3ct, 2017-01-29:
[My setup - Ubuntu 14.04, python 2.7.6, Dell XPS 9343, Artisan 1.0.0 4 downloaded & installed .deb as per [instructions](https://github.com/artisan-roaster-scope/artisan/blob/master/wiki/Installation.md)]

Hi there, I am trying to get a proper callprogram() implementation running with artisan.  I'm running into issues because my python script/package uses the logging package and sets up some substantial file logging.  As a result, Artisan does not read anything other than "" in its `output = p.communicate()[0]` call.  

I first verified that I could write a super-simple script that would work in artisan, and started adding to it, the first instinct being to set up logging to a file to see what is going on.  That seems to screw up the stdout for the script somehow, even if the script runs as intended from Terminal.

Here's what works for Artisan, using a logging.StreamHandler() (because, stderr is going into a black hole, no logging, as expected):

````
#!/usr/bin/env python

import sys
import logging
import logging.handlers

logger = logging.getLogger(__name__)
logger.setLevel(logging.DEBUG)
# handler = logging.handlers.RotatingFileHandler(
#     'read-et-f.log', mode='a', maxBytes=1000000, backupCount=10)
handler = logging.StreamHandler()
# create formatter
formatter = logging.Formatter('%(asctime)s - %(name)s - line %(lineno)d - %(levelname)s - %(message)s')
# add formatter to handler
handler.setFormatter(formatter)
# add handler to logger
logger.addHandler(handler)

logger.info("before print")
print("100.0,95.0")
logger.info("after print")
````

And, switching to a logging.handlers.RotatingFileHandler causes subprocess.Popen() in callprogram() to fail to see the output of the print() statement:

````
#!/usr/bin/env python

import sys
import logging
import logging.handlers

logger = logging.getLogger(__name__)
logger.setLevel(logging.DEBUG)
handler = logging.handlers.RotatingFileHandler(
    'read-et-f.log', mode='a', maxBytes=1000000, backupCount=10)
# handler = logging.StreamHandler()
# create formatter
formatter = logging.Formatter('%(asctime)s - %(name)s - line %(lineno)d - %(levelname)s - %(message)s')
# add formatter to handler
handler.setFormatter(formatter)
# add handler to logger
logger.addHandler(handler)

logger.info("before print")
print("100.0,95.0")
logger.info("after print")
````

Are there any recommendations for logging to file in artisan python scripts destined to be used with callprogram()?

I wrote a "pretend-artisan.py" subprocess.Popen() call script, which does work properly and allows logging to file.  So, I'm assuming the 'magic' is in the 'env=aw.calc_env()' call within callprogram(), that sets something up in a way that file logging fails?

Test files: 
[artisan-callprogram-test.tar.gz](https://github.com/artisan-roaster-scope/artisan/files/737783/artisan-callprogram-test.tar.gz)

roundfile, 2017-01-31:
For what it is worth, your code "read-et-f.py" using the logging.handlers.RotatingFileHandler() works as expected with Artisan v1.0.0.4 and Python 2.7.12 on my Windows 7 computer.  Artisan sees the values from the print() statement and the log file is properly written.

int3ll3ct, 2017-02-01:
@roundfile , thanks for the info, reassuring that it's reliable elsewhere.  I've set up without logging for now on my system, and I turn logging on when verifying scripts manually to work out problems.  I'll write this on off as 'OS/system idiosyncrasies.'

closed 2017-02-01 by int3ll3ct

MAKOMO, 2017-02-04:
There was an improvement of the program call function for the Linux platforms committed around 11/2016 after the v1.0.0b4 release. That could resolve your issue. Did you try to to run from trunk source?

int3ll3ct, 2017-02-06:
I did not run from source. I have terrible luck with all things QT. I promise to attempt that once I get through some more pressing matters elsewhere. I had to disable file logging which then allowed my script to work, meanwhile. Thanks for commenting.