artisan-roaster-scope/artisan #116: "Modbus Error: readSingleRegister() Exception response @line 28941"

pantelis1300, 2017-02-16:
Hi guys,

I am trying to connect an FZ-94 bought in November 2016 with a Windows 10 laptop but I receive the error message above. Any friendly advice please is welcome, I am desperate. Thanks in advance.

Best Regards

Pad

pantelis1300, 2017-02-16:
UPDATE:

I have installed Artisan version 0.9.9 (the previous version that returned error 28941 was Artisan 1.0.0) and now I get a new error message "Modbus Error: readSingleRegister() Exception response @line 27248". So yes, the error code changed from 28941 to 27248 when i moved from Artisan 1.0.0 back to Artisan 0.9.9.

Thanks

Pad

pantelis1300, 2017-02-16:
UPDATE 2:

I somehow  managed to make it work. I changed the baud rate to 19200 and Timeout to 1 and it seems it is working. The error message (27248) persists though - not sure what it means and if it is affecting anything. Any suggestions?

Thanks

Pad

poundy, 2017-02-25:
its not an error code, its a line number in the code. It's telling you where it stopped.  

As that error is going to be very specific to the device and roaster, you might not get too much assistance from people or posts that don't have that specific setup.  The good news is that FZ-94 and Artisan is a well supported scenario, I'd suggest you check out the related posts at https://artisan-roasterscope.blogspot.com.au/2016/08/fz-94-1-bumping-up-modbus.html

MAKOMO, 2017-02-26:
The error message indicates simply a read error. You might have asked Artisan to read data from a MODBUS register or MODBUS slave that is not accessible. As this seems to be a configuration issue and not point to an issue in Artisan I close this one.

closed 2017-02-26 by MAKOMO

pantelis1300, 2017-02-27:
Hi guys,

yes, the error actually did not appear at all the last time I tried it. It seems to pop up randomly from time to time.

I will try "bumping up" modbus when I will have some free time.

Thanks a lot for your comments, I appreciate this.

Pad