artisan-roaster-scope/artisan #117: Artisan 1.0.0 stops tracking when I click the "Charge" button

cruiten, 2017-02-25:
Artisan 1.0.0 stops tracking when I click the "Charge" button and enter a BT value of 320. I currently do not have any thermocouples connected to my computer and I am manually entering the temperatures that I read from my multimeter that is connected to a K-type thermocouple. Under Config > Device Assignment I am using input device "NONE". This has been working fine with Artisan 0.9.9.

I must be configuring something incorrectly, because when I do a Factory Reset, it is working as expected.

When it stops tracking, I see the following in the Artisan Error Log:
```
version = 1.0.0

1 15:50:24.549 Exception: updategraphics() list index out of range@line 1996
```

Artisan Platform data:
```
version = 1.0.0 (6)
Architecture = ('32bit', 'WindowsPE')
Machine = AMD64
Platform name = Windows-7-6.1.7601-SP1
Processor = Intel64 Family 6 Model 42 Stepping 7, GenuineIntel
Python Branch = 
Python Build = ('v2.7.13:a06454b1afa1', 'Dec 17 2016 20:42:59')
Python Compiler = MSC v.1500 32 bit (Intel)
Python Implementation = CPython
Python Revision = 
Python version = 2.7.13
Release = 7
System = Windows
Version = 6.1.7601
Win32 = ('7', '6.1.7601', 'SP1', u'Multiprocessor Free')
```

I have attached the Settings file that I was using when this problem occurred.
[artisan-settingsv1_0_0.zip](https://github.com/artisan-roaster-scope/artisan/files/800761/artisan-settingsv1_0_0.zip)

cruiten, 2017-02-25:
It appears that this is related to the whether I have the Curves "ET" checkbox selected or not on the "ET/BT" tab in Config > Device Assignment.

If I un-check Curves "ET", hit "OK", and exit out of Artisan, then I always encounter the reported problem the next time that I use Artisan.

However, if Curves "ET" is checked, then this problem does not manifest itself.

Incidentally, if I do a run with the "ET" checkbox checked, and then un-check it then the following run will work just fine.

It is almost as if the Bean Temperature variable is not properly initialized when the ET checkbox is not selected.

MAKOMO, 2017-02-25:
Wow! Thanks for the detailed report.  I will look into it after my vacation. Seems to be a small thing to fix.

MAKOMO, 2017-02-26:
Fixed by [Commit 61db322](https://github.com/artisan-roaster-scope/artisan/commit/61db32229a2f1f44e380e8c28b31d9b9aace7cd3)

closed 2017-02-26 by MAKOMO