artisan-roaster-scope/artisan #120: Bad ET and BT temp readings

redadder22, 2017-06-19:
Hi,

I've just installed two J type thermocouples onto my gas Cafemino roaster via Phidgets 1048 and the readings seem normal when the roaster is cool but as the gas is turned on and up the temps seem to rocket to almost twice the reading of the factory made temp sensor. I've looked at all the settings and they seem normal. Any suggestions?

MAKOMO, 2017-06-19:
Under menu Config >> Device, 4th tab "Phidgets", you have to indicate the probe type for your 1048. k-type is the default. Set this to j-type.

M

redadder22, 2017-06-19:
Hi,

I already did this on Phidgets and Artisan but still the same problem.

MAKOMO, 2017-06-19:
Oh, wow! Just looked at the code and saw this

        self.phidget1048_types = [
            ThermocoupleType.PHIDGET_TEMPERATURE_SENSOR_K_TYPE,
            ThermocoupleType.PHIDGET_TEMPERATURE_SENSOR_K_TYPE,
            ThermocoupleType.PHIDGET_TEMPERATURE_SENSOR_K_TYPE,
            ThermocoupleType.PHIDGET_TEMPERATURE_SENSOR_K_TYPE] # probe types (ThermocoupleType)

of course it should read

        self.phidget1048_types = [
            ThermocoupleType.PHIDGET_TEMPERATURE_SENSOR_K_TYPE,
            ThermocoupleType.PHIDGET_TEMPERATURE_SENSOR_J_TYPE,
            ThermocoupleType.PHIDGET_TEMPERATURE_SENSOR_E_TYPE,
            ThermocoupleType.PHIDGET_TEMPERATURE_SENSOR_T_TYPE] # probe types (ThermocoupleType)

Wonder why this was not discovered some years before. I'll fix this for you and send you a new build. 
Which platform are you on? Windows, Mac, Linux?

Thanks for your report,
Marko

redadder22, 2017-06-19:
Hi Marko,

Ok thanks- I'm suing Windows.

redadder22, 2017-06-19:
I meant using!!

MAKOMO, 2017-06-19:
I have a new Windows build ready, but cannot contact you directly via Github. Could you please contact me via the Artisan blog contact form at https://artisan-roasterscope.blogspot.de/p/contact-me.html such that I can send you the link to the build via a direct email? Thanks.

MAKOMO, 2017-06-20:
It turned out that the original code was absolutely fine (just setting the default types to k-type). The issue must be hardware related as a verification revealed that probe types are correctly set by Artisan in a Phidgets 1048.

closed 2017-06-20 by MAKOMO