artisan-roaster-scope/artisan #121: Background profile settings disappear when ok is selected

SteveCoss, 2017-06-19:
Hi there,
First off, THANKS so much for developing this.

I am having an issue where background profiles are plotting with with line size zero, then when I change apply and select OK, the setting revert. I am on windows 7

MAKOMO, 2017-06-19:
Dear Steve,

could you please explain your issue a bit more in detail? The line size of your background curves is set to zero and you cannot change it to any other value? How did Artisan came into this state? Originally, curves have a line size of 1 or 2, right? Which background curves are you talking about? Which menus/dialogs are you using to change the line size? Could you provide screenshots of the graph and the dialogs you are operating? I just tried to replicate your issue, but failed on my Mac (I have no Windows 7 available).

SteveCoss, 2017-06-19:
Absolutely, sorry to make you ask.

Basically, whenever I load a background profile (under roast tab), the events are not visible. If I press the line display button and find the event lines, there are no markers set and the line width is zero. I can change them, and the lines shoe up after I hit apply, but when I click OK they disappear again. I am going to be at my roasting

computer in a few min and can send screen shots.

Cheers,

Steve

SteveCoss, 2017-06-19:
Image 1 is how I am loading a background profile. Two shows how it loads up without any event lines showing. Three is the button I am pressing to change line display. Four shows the default desplay for power (the name won't stay set either). Five shows the changes working after I press apply. 6 shows how it looks after I hit okay.

Thanks!
________________________________
From: Coss, Stephen P. <coss.31@buckeyemail.osu.edu>
Sent: Monday, June 19, 2017 3:00:43 PM
To: artisan-roaster-scope/artisan; artisan-roaster-scope/artisan
Cc: SteveCoss; Author
Subject: Re: [artisan-roaster-scope/artisan] Background profile settings disappear when ok is selected (#121)

Absolutely, sorry to make you ask.

Basically, whenever I load a background profile (under roast tab), the events are not visible. If I press the line display button and find the event lines, there are no markers set and the line width is zero. I can change them, and the lines shoe up after I hit apply, but when I click OK they disappear again. I am going to be at my roasting

computer in a few min and can send screen shots.

Cheers,

Steve

MAKOMO, 2017-06-20:
Steve, there were no images attached. You need to attach images directly on Github by "dragging & dropping or selecting them" in your Web browser. You can see those images being rendered on clicking the "Preview" tab. Replying with images to the emails seems not to work.

Anyhow, it is still unclear (sorry!) what you refer to by "press the line display button". Which dialog, which menu item and/or which button are you operating here?

I assume that you are aware that you have to tick the "Events" flag in the background dialog to have the events of the background profile be displayed (as marked in the pic below).

[image]

SteveCoss, 2017-06-20:
[image]
[image]
[image]
[image]
[image]
[image]
Okay, so there's the images. I am selecting the events 'tick'.

MAKOMO, 2017-06-20:
Dear Steve,

I now have a clear idea what is going on, but still cannot reproduce the effect. It seems that your Artisan configuration is somehow "broken". One indication is the "Background_line7" label in the dialog which should instead indicate the proper event name as set by the user in Artisan. One complication is that this line/curve configuration/dialog is generated/inherited by the underlying graph lib and not part of the Artisan code. I might be able to work around this issue for you by some code changes, but this is only possible if I can reproduce the issue. To enable me to do this, please send me your Artisan configuration file (save via menu Help >> Save Settings) and this profile as alog file per email. As I don't want to disclose my email here to spammers, please use the Contact form on the Artisan blog <https://artisan-roasterscope.blogspot.de/p/contact-me.html> to send me a message (that will indicate your email address to me) and I will reply by email such that you can send me the two files. 

Thanks for your support on this. 
Marko

SteveCoss, 2017-06-21:
> blogspot.de/p/contact-me.html to send me a message (that will indicate

Hey Marko,
Just roasted some and my active roast event weren't plotting. I went into
the event settings, and the event type (power, fan etc.) Pull down menu no
longer has those items either. Just line_7 line_6 and so on.

Cheers,
Steve

MAKOMO, 2017-07-03:
Turned out to be caused by messed up application settings.

closed 2017-07-03 by MAKOMO