artisan-roaster-scope/artisan #129: Describe the packaging process

baderas, 2017-09-02:
After installing everything as described in https://github.com/artisan-roaster-scope/artisan/blob/master/src/setup-packages-linux.txt and spending hours in figuring out how to install everything correctly and create .deb packages, I did not succeed. I get .deb packages using build-ubuntu.sh and build-linux.sh but they miss Phidgets22 and the content looks quite different from the .deb packages that are downloadable (a lot more egg folders).

Can you explain how the ubuntu/debian packages are created? (everything is installed correctly).

MAKOMO, 2017-09-02:
Dear Andreas,

building Artisan on any platform is quite difficult due to the large number of components involved that constantly change. My Linux build setup is especially complex and involves two (outdated) CentOS VMs (32bit/64bit) as well as a Ubuntu VM to generate the packages. The Phidgets22 support is not yet stable and the Linux (incl. RPi) build setups are not up to date regarding those yet. My plan is to completely redo the Linux build setups and just use one 64bit CentOS 7.2 VM for this and moving also the Linux part over to Python v3, the latest Qt 5.9 and the current pyinstaller instead the now unsupported bbfreeze. So I will not describe the current outdated setup and instead come up with a renewed and hopefully simplified setup that can be better supported and documented. Just have to find the time for this...

PS: Thanks for your pull. I will have a look on this the next days.

9b, 2017-09-30:
Will you be cleaning up the Mac build process as well? Similar to @baderas, I spent a couple hours getting everything in place only to find that some files did not appear to copy or link on the output DMG. I'd happily contribute development efforts to the project if I can get the build process up and running.

MAKOMO, 2017-10-19:
There are three build scripts for Mac OS X. Two are based on py2app (build-mac.sh for Python v2.7 and build-mac35.sh for Python 3.5) and one based on pyinstaller on Python3.5 (build-mac35-pi.sh). Those scripts work for me without any issue and those two tools and 3 scripts are responsible to copy all relevant libs and other components.

closed 2017-10-19 by MAKOMO

jasonwardenburg, 2018-04-08:
Hi,
can one or two of you guys describe a working build setup? Are you guys working on linux or windows?

RickGroszkiewicz, 2018-04-08:
AFAIK, there is no standard "build setup", due to the numerous python packages required. I would love to know the answer too, so I could help debug some of the problems that I have run into.

rpaulo, 2018-04-08:
@RickGroszkiewicz see my answer here https://github.com/artisan-roaster-scope/artisan/issues/174

RickGroszkiewicz, 2018-04-09:
I was thinking in terms of building artisan for Linux. Are there any guidelines for that environment?

rpaulo, 2018-04-10:
There's the setup-packages-linux.txt, build-centos-pi.sh and build-linux.sh files

RickGroszkiewicz, 2018-04-10:
Thanks, I will review these files when I get some free time.