artisan-roaster-scope/artisan #130: v1.1.0 64 bit version not working

frankc60, 2017-09-02:
I'm downloaded the latest version "artisan-linux-1.1.0_amd64.deb", and installed it. When I run it from the command line, it reports "Segmentation fault (core dumped)", which is where the program tried to access a memory area out side its own section. OS blocks it.
I got the 32 bit version working (artisan-linux-1.1.0_i386.deb), but can't access my Hottop FTDI virtual usb connection. By the way, my OS is 64bit (linux) Mate 1.16.0, and I did have v0.9.8 working a while back okay.

How have you compiled this, if I download (or pull) the source is it straightforward to build myself ?
Thanks, Frank

baderas, 2017-09-03:
I tried to create new packages and spend more than a day with several Ubuntu versions and achieved only a working 32bit package of artisan. See https://github.com/artisan-roaster-scope/artisan/issues/129 for further details.

Instead of creating a new package, just download the source and execute `python2 artisan.py`. You will need to install several dependencies with pip (or apt-get) and you will only have english langugage. But this works quite easily.

frankc60, 2017-09-04:
Thanks baderas, I have started, but will take a bit to get all the dependencies installed

closed 2017-09-04 by frankc60