artisan-roaster-scope/artisan #132: Output report does not function

9b, 2017-09-29:
When trying to export a report of the roast, I am getting an error that appears to use a hardcoded path. Is there any way to configure this?

Version: Artisan 1.1.0 (2)
Platform: Mac OS X (10.2.6)

Preview of what I see:
[image]

MAKOMO, 2017-09-30:
Artisan is generating the report as a html page in a temporary file automatically selected by an underlying library in cooperation with the OS. Artisan has no influence on its name/location. However, here the generation of that html report seems to fail and therefore that file does not exist. Could you please share the alog file that causes this problem such that I can analyse where things break (assuming your OS is 10.12.6 macOS Sierra and not 10.2.6, right)?

9b, 2017-09-30:
Thanks for the quick response. I've attached my alog file for analysis and you are correct, it's macOS Sierra. 

[Ethiopia Nano Challa_17-09-29_1224.alog.zip](https://github.com/artisan-roaster-scope/artisan/files/1346713/Ethiopia.Nano.Challa_17-09-29_1224.alog.zip)

myTonino, 2017-09-30:
Hm. This works as expected from current trunk as well as from the v1.1 build on my macOS Sierra 10.12.6. There must be an issue on your system. Any related messages in the Console log? Do you have this issue with any profile (also the empty one)?  Are the ranking and batch Web reports working for you?

9b, 2017-09-30:
Strange. I just fired up the build (non-sudo) and it worked on blank and with my alog. I think at some point I opened the app as root because I was having issues connecting, though that later wasn't found to be the issue. Thanks for confirming it worked on your side and making me think to try it.

closed 2017-09-30 by 9b