artisan-roaster-scope/artisan #133: Event replay on BT out of sync with background

mbenedet, 2017-10-21:
Hi Marko - I turned on Playback Events "by BT" on the Background dialog with playback warning.  It does indeed give warning but not in sync with the background BT.

Similarly in the Events dialog, "Show on BT" doesn't seem to do anything.  All my Events shows by Type still drop flags to the ET curve.

On my first attempt with these new settings, Artisan also forced my Power slider to move to the background settings and marked those events on the curve without me having made those changes.

Thank you!

mbenedet, 2017-10-21:
Perhaps I misunderstood.  "Playback" forces the events to occur.  "Playback aid" is probably all I need if I'm roasting manually?

MAKOMO, 2017-10-22:
Yep, "Playback" is a playback function, re-running the events based on the background profile (time or temperature) in the actual roast. "Playback aid" is only giving you warning notices the selected seconds before the event occurred in the background profile.

M

closed 2017-10-22 by MAKOMO