artisan-roaster-scope/artisan #136: Artisan doesn't work on Debian

rpaulo, 2017-10-28:
Looks like the bundled libraries don't work on Debian:

{{{
% /usr/bin/env LD_LIBRARY_PATH=/usr/share/artisan:$LD_LIBRARY_PATH LIBOVERLAY_SCROLLBAR=0 GTK2_RC_FILES="$HOME/.gtkrc-2.0"  gdb artisan
gdb: /usr/share/artisan/libtinfo.so.5: no version information available (required by gdb)
gdb: /usr/share/artisan/libtinfo.so.5: no version information available (required by /lib/x86_64-linux-gnu/libreadline.so.7)
GNU gdb (Debian 7.12-6+b1) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from artisan...(no debugging symbols found)...done.
(gdb) r
Starting program: /usr/share/artisan/artisan 
/bin/zsh: /usr/share/artisan/libtinfo.so.5: no version information available (required by /bin/zsh)
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe4e48700 (LWP 10396)]

Thread 1 "artisan" received signal SIGSEGV, Segmentation fault.
0x00007ffff7de4d3c in ?? () from /lib64/ld-linux-x86-64.so.2
(gdb) bt
#0  0x00007ffff7de4d3c in ?? () from /lib64/ld-linux-x86-64.so.2
#1  0x00007ffff7decee1 in ?? () from /lib64/ld-linux-x86-64.so.2
#2  0x00007ffff7de8744 in ?? () from /lib64/ld-linux-x86-64.so.2
#3  0x00007ffff7dec709 in ?? () from /lib64/ld-linux-x86-64.so.2
#4  0x00007ffff79b8ee9 in ?? () from /lib/x86_64-linux-gnu/libdl.so.2
#5  0x00007ffff7de8744 in ?? () from /lib64/ld-linux-x86-64.so.2
#6  0x00007ffff79b9531 in ?? () from /lib/x86_64-linux-gnu/libdl.so.2
#7  0x00007ffff79b8f82 in dlopen () from /lib/x86_64-linux-gnu/libdl.so.2
#8  0x00000038a46398e8 in ?? () from /usr/share/artisan/libGL.so.1
#9  0x00000038a46390e9 in ?? () from /usr/share/artisan/libGL.so.1
#10 0x00000038a461c408 in ?? () from /usr/share/artisan/libGL.so.1
#11 0x00000038a4618237 in glXGetFBConfigs () from /usr/share/artisan/libGL.so.1
#12 0x00000038a4618d03 in glXChooseFBConfigSGIX () from /usr/share/artisan/libGL.so.1
#13 0x00007fffe5e24e5e in ?? () from /usr/share/artisan/Resources/qt_plugins/platforms/libqxcb.so
#14 0x00007fffe5e25093 in ?? () from /usr/share/artisan/Resources/qt_plugins/platforms/libqxcb.so
#15 0x00007fffe5dc7d3d in ?? () from /usr/share/artisan/Resources/qt_plugins/platforms/libqxcb.so
#16 0x00007fffe5db8da9 in ?? () from /usr/share/artisan/Resources/qt_plugins/platforms/libqxcb.so
#17 0x00007ffff4f8a863 in QWindowPrivate::create(bool) () from /usr/share/artisan/libQt5Gui.so.5
#18 0x00007ffff58591c4 in QWidgetPrivate::create_sys(unsigned long long, bool, bool) () from /usr/share/artisan/libQt5Widgets.so.5
#19 0x00007ffff5859695 in QWidget::create(unsigned long long, bool, bool) () from /usr/share/artisan/libQt5Widgets.so.5
#20 0x00007ffff585cb0a in QWidget::setVisible(bool) () from /usr/share/artisan/libQt5Widgets.so.5
#21 0x00007ffff62c6774 in ?? () from /usr/share/artisan/PyQt5.QtWidgets.so
#22 0x00007ffff6235428 in ?? () from /usr/share/artisan/PyQt5.QtWidgets.so
#23 0x000000000047086a in PyEval_EvalFrameEx ()
#24 0x0000000000471122 in PyEval_EvalFrameEx ()
#25 0x0000000000472687 in PyEval_EvalCodeEx ()
#26 0x0000000000472792 in PyEval_EvalCode ()
#27 0x0000000000471113 in PyEval_EvalFrameEx ()
#28 0x0000000000472687 in PyEval_EvalCodeEx ()
#29 0x0000000000472792 in PyEval_EvalCode ()
#30 0x0000000000471113 in PyEval_EvalFrameEx ()
#31 0x0000000000472687 in PyEval_EvalCodeEx ()
#32 0x0000000000472792 in PyEval_EvalCode ()
#33 0x0000000000493c9b in PyRun_StringFlags ()
#34 0x0000000000414344 in main ()
(gdb) 
}}}

MAKOMO, 2017-10-29:
The actual 1.2b deb build runs fine on my Ubuntu 16.04. Which OS you are running?

rpaulo, 2017-10-29:
Debian testing

MAKOMO, 2017-10-29:
That might be either too new, or have a problem running the Qt5.9 contained in Artisan

sergeibux, 2017-11-01:
I'm running a stable Debian version and I've installed this with no problem. So I think it should come from your debian maybe.

arraylabs, 2017-11-05:
How well does artisan run on the pi 3? Looking to integrate it into custom built roster.

MAKOMO, 2017-11-09:
As confirmed by several people, the current build of v1.2 installs and runs well on standard Debian installations, so I close this one.

closed 2017-11-09 by MAKOMO