artisan-roaster-scope/artisan #137: Mastech 6514 and Debian

sergeibux, 2017-11-01:
Hi,

I run Artisan since a long time now. Today, I've tried to install v1.2.0b3_amd64.
Install was successful with no error.
I've had dialout's group to my user.
My problem is a message from Artisan that tell me it's receiving 35 bytes in order 18.

> "mastech6514temperature(): 35 bytes received but 18 needed"
I didn't found any solution.

MAKOMO, 2017-11-09:
The Mastech adapter code seems not to be Python3 compatible, but Artisan is based on Python3 from v1.2 onwards. I might be able to fix this (although it is not my code and I have no specification of the underlying protocol), but I have no device to test this. Could you support me on testing this? If yes, please write me a message via the Artisan blogs contact formula at https://artisan-roasterscope.blogspot.de/p/contact-me.html. Thanks!

sergeibux, 2017-11-13:
I've tried to install from src the new version on my Debian but I don't know how to do that. Coul'd you please show me the way to do it ?
Thanks a lot.

yaleforks, 2017-12-11:
I have the same issue on Win10 x64. Let me know what I can do to help

MAKOMO, 2017-12-11:
Which "same" issue do you have? The Mastech 6514 is not working for you either with the Windows 1.2b?

yaleforks, 2017-12-11:
I get this error on 1.2b

"mastech6514temperature(): 35 bytes received but 18 needed"

Downgrading to 1.1 works for me but 1.2b shows the error

MAKOMO, 2017-12-11:
Contact me via the online form here https://artisan-roasterscope.blogspot.de/p/contact-me.html and I will build something for you to test that hopefully resolves this. This is a good timing to get this resolved before the final v1.2 release. Thanks for your support!

MAKOMO, 2017-12-12:
As yaleforks confirmed that my fix works I close this issue.

closed 2017-12-12 by MAKOMO

sergeibux, 2017-12-30:
OK, I'm sorry, I didn't found the time to try it.
Thanks for the job.