artisan-roaster-scope/artisan #138: Windows 10 32bit OS

jessemedley, 2017-11-16:
artisan-win-1.2.0b3 installs but does not run on my windows 10 32-bit operating system. It says "this app can't run on your pc." when I try to start it.

I have both phidgets 21 and 22 installed and Microsoft Visual C++ 2008 SP1 Redistributable Package (x86) is installed.

MAKOMO, 2017-11-16:
As indicated in the version history, 32bit Windows is not support by Artisan v1.2 and newer.

closed 2017-11-16 by MAKOMO

jessemedley, 2017-11-17:
Sorry, I don't see it listed in the version history.
here: https://github.com/artisan-roaster-scope/artisan/blob/master/wiki/ReleaseHistory.md
Is there a version history somewhere else?

MAKOMO, 2017-11-17:
It is noted on the version history summary on the main page for v1.1 as being the last version supporting 32bit windows. I'll add a note on the detailed version history too and will try to ensure that the installer works only for 64bit systems. Thanks for reporting.

MAKOMO, 2017-11-17:
Actually, the 32bit/64bit discussion is mostly about the builds as the source code supports both version. I am just not providing any 32bit builds for any OS. However, you or someone else could do so. Artisan is open-source after all!