artisan-roaster-scope/artisan #140: dropdown menues not readable

benoitmathieu, 2017-12-19:
Hi Marko and Rafael
In the beta version 1.2.0B3, the dropdown menues related to "Change" in Config>Device>Phidgets are not readable upon dropdown. So you have to choose at random.
Almost all "change" lines are showing this problem in this tab.

benoitmathieu, 2017-12-19:
[image]

benoitmathieu, 2017-12-19:
[image]

MAKOMO, 2017-12-20:
Dear Mathieu, thanks for this report. Seems to be only an issue on Windows. I just fixed this and will roll this in the v1.2 builds to be released tomorrow.

closed 2017-12-20 by MAKOMO