artisan-roaster-scope/artisan #141: V1.2.0B3 : support to Phidgets OUT1100

benoitmathieu, 2017-12-20:
Hello @MAKOMO ,

I have tried Artisan 1.2.0B3 with new Phidgets TPM1101 (4xTC) and OUT1100 (4xPWM) with Phidget22 library/drivers both connected to HUB0000

TCs are working fine when I setup "Meter" to [Phidget TMP11014xTC] I could have both readings and graph curves for BT and ET when working with two K thermocouples - That's perfect!

Unfortunately, I do not see how to setup regulation (internal Artisan PID) output out on the OUT1100 module. This is the missing link to get full management over the electric heater (through SSR) using Phidgets.

Another bonus would be to also control the fan power through anoth port on the OUT1100 using cursor output redirection.

Could you tell me if this is already doable in the current state of the 1.2.0 release or if this requires extra development.

In the latter case I would be happy to contribute if some guidance is available.

Cheers,

Benoit

MAKOMO, 2017-12-20:
Dear Benoit, OUT1100 is already implemented. Thus I am not sure what you refer to by "cursor output redirection". The OUT1100 can be controlled via custom buttons or sliders and sliders can be moved by the internal PID output. As "Action" you configure "PWM Command" and as command given in the "Documentation" you use out(n,v) where n is the channel of the OUT1100 you want to address, 0-3, and v is the PWM value you want that channel to be set to (0-100%). Might be that the OUT1100 control is still broken in the v1.2b, but it works for the final version v1.2 to be released tomorrow.

closed 2017-12-20 by MAKOMO

benoitmathieu, 2017-12-21:
Thanks for these pieces of information, Marko.
I now have it working fine and have catched up with Artisan (yours? :-) ) logic.
For my personal elevation, is there a repository where all these commands are documented?
In the "Documentation" I could find bits an pieces, but that's quite spread al over the topics.
I suppose everything is in the code, but a command wiki would be handy.
Thanks again for your great support.