artisan-roaster-scope/artisan #145: Issue with statistics/events

fraklo, 2017-12-29:
Running on Windows 10 64 bit
Version 1.2.0 (3)

I am having an issue where my statistics are showing all screwed up after a roast.
[image]

The event times don't match up with the data:
[image]

I've tried updating the values for the events using the properties dialog but it doesn't affect anything and when I open it back up the values have reset.

The one thing I did notice is that there doesn't seem to be a CHARGE event recorded in my last few roasts. Thinking this may have thrown things out of whack because things have been working perfectly fine up until the latest update or two. (Not quite sure when the problem started).

Much thanks for any help/insights!

MAKOMO, 2017-12-31:
Without a proper CHARGE event those phases cannot be calculated correctly. Just add a CHARGE event to this profile by a right-click on the BT at the corresponding position and select CHARGE from the popup. This should fix this one. I close this issue as it is not a but in the software.

closed 2017-12-31 by MAKOMO

fraklo, 2017-12-31:
Mind blown. Was not aware about the right clicking to add events bit! I just tried it and was able to correct a previous log.

Also, just roasted and instead of relying on the auto charge event we clicked the charge event at the beginning and all went perfectly fine. Thanks for your time and doubly thanks for the awesome work on this! 🙇