artisan-roaster-scope/artisan #146: Crash after editing roast event

RickGroszkiewicz, 2017-12-31:
After completing a roast, I typically press OFF, then A to autosave the roast results. The final step is to manually edit the events to get rid of those I consider extraneous (or duplicate) and autosave again. 

I downloaded and installed version 1.2 under Linux Mint. In my roasts today, I can not delete any events. As soon as I click "delete", artisan closes immediately.

MAKOMO, 2017-12-31:
I cannot reproduce this one. Tested under CentOS 7 for a lack of Linux Mint.

RickGroszkiewicz, 2017-12-31:
I recorded a shot video of the problem - you should turn the sound WAY down ...
https://youtu.be/uVn2_sJ6EJ0

MAKOMO, 2017-12-31:
Seems to be a build/lib issue for your platform. Could you try to start Artisan via 

# /usr/share/artisan/artisan

from a shell? In the best case this would print out a message on the crash that could help to diagnose the issue.

RickGroszkiewicz, 2017-12-31:
Terminal display gives "segmentation fault" when artisan disappears.

RickGroszkiewicz, 2017-12-31:
OOPS, pressed wrong button!

MAKOMO, 2017-12-31:
Hm. Not very informative this message. There must be a conflict between your OS and Artisan. Wonder why the crash exactly happens at this delete event action. Just checked the source code of this action and couldn't spot anything specific. Could you share your Artisan settings file? Alternatively you could save your settings (for later reconstruction), make a factory reset and see if the crash still happens. Reload your settings file to restore your setup. If the crash does not happen directly after the factory reset, there might be something wrong in the settings. If it is still crashing I am running out of ideas...

RickGroszkiewicz, 2018-01-01:
I saved my settings, then tried a factory reset. Artisan acts exactly the same, and gives the same  "segmentation fault" when artisan disappears. Here is my settings file (renamed to .txt):
[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/1595589/artisan-settings.txt)

RickGroszkiewicz, 2018-01-07:
As an additional test, I restored a backup from June 2017. After taking care of all the subsequent Linux "system updates", I upgraded artisan to version 1.2.0. It acts the same, crashing when I delete an event for a completed roast profile. I also duplicated this behavior using the Cinnamon desktop environment, as well as Cinnamon with software rendering.

rpaulo, 2018-04-04:
Please try the latest build from here https://dl.bintray.com/artisan/artisan-artifacts/

raescuin, 2018-04-04:
> just tried this out and I'm not getting a BT reading. Also it seems kinda
> slow on everything..
>
> Robert
>
> --- Robert

is this the latest build?

--- Robert

RickGroszkiewicz, 2018-04-05:
Robert, that link is the latest pre-release 1.3.0 version or artisan. Based on my experience, the pre-release version GUI seems slower than the final released version.

I will download this and give it a try when I roast this weekend.

rpaulo, 2018-04-05:
@RickGroszkiewicz please file another issue if the pre-release builds are too slow.  There's no special reason why they should be slower.

rpaulo, 2018-04-08:
@RickGroszkiewicz did you try the latest builds?  We'd like to know if this is still a problem.

RickGroszkiewicz, 2018-04-09:
Sorry, my schedule simply did not allow for roasting yesterday. I will try out a "dummy roast" this evening to see how it behaves.

raescuin, 2018-04-09:
fyi, I tried it yesterday 3 roasts and it went smoothly with no problems of whatsoever.

MAKOMO, 2018-04-09:
@RickGroszkiewicz: Thanks for your support! Please save your settings using the Artisan v1.2 before installing and starting v1.3 such that we have a reference point to track back if things fail. One user just reported a crasher on Linux on opening the Extra Devices tab, pressing delete after pressing add. We cannot reproduce this yet. Would have time to test this action and if it crashes send use the full crash log that is produced by our new crash logger. Thanks!

MAKOMO, 2018-04-09:
@raescuin : thanks for this positive report!

RickGroszkiewicz, 2018-04-09:
Marko, I already deleted 1.2.0 and installed 1.3.0. However, I did create a full backup before making any changes to artisan (takes 8 minutes to restore from backup).

MAKOMO, 2018-04-10:
@RickGroszkiewicz : Thanks for this excellent explanation (again) in your video and again I couldn't reproduce this on neither Windows, Mac OS X nor CentOS. However, a clean Ubuntu 16.04 showed the exact same crasher. So I could investigate today and found that this happens on clearing the selection of a QTable. This must be a Qt conflict with some Linux desktops. I found away around this and fixed the same issue also in other tables like the buttons list, the alarm table and some more. Could you please test a current build on your setup and confirm that my patch also fixes this for you? Thanks Rick!

MAKOMO, 2018-04-10:
Wow! Thanks for your fast reaction and testing work. Then we can finally tick off this one and go forward with the release. Again thanks for your support, Marko

closed 2018-04-10 by MAKOMO