artisan-roaster-scope/artisan #147: Background profile missing ROR

RickGroszkiewicz, 2017-12-31:
I downloaded and installed version 1.2.0(3) under Linux Mint with the LXDE environment. I noticed today that the DeltaBT curve is not shown in my background profile. Those profiles were created using artisan version 1.1.0.

MAKOMO, 2017-12-31:
Do you have DeltaBT ticked in the background dialog (menu Roast >> Background)? If yes, and you still do not see the DeltaBT of the background profile, please share that alog file.

RickGroszkiewicz, 2017-12-31:
Yes, I also tried turning that setting on and off during the roast, but it made no difference.
[Background ROR problem.zip](https://github.com/artisan-roaster-scope/artisan/files/1595370/Background.ROR.problem.zip)

MAKOMO, 2017-12-31:
Hm. They look beautiful as background profile  incl. DeltaBT on my Mac. Something might be different. Could you also share your Artisan settings file (save via menu Help >> Save Setting) to further investigate? I don't have Linux Mint, but I will also check the situation on my build CentOS VM, however, I would be surprised if this is a platform dependent issue.

RickGroszkiewicz, 2017-12-31:
I checked in both Windows 7 and Linux Mint. The background ROR curve disappears when you press the "start" button.

MAKOMO, 2017-12-31:
Just checked again, but for me the background RoR curve stays where it belongs also after pressing START. Could you share your Artisan settings file please?

RickGroszkiewicz, 2018-01-01:
Here is my settings file: 
[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/1595580/artisan-settings.txt)

MAKOMO, 2018-01-02:
You have the RoR filter activated (menu Tools >> Extras, 2nd tab Filters, RoR Filter section).

[image]

Just untick that one and your BT RoR should display fine.

RickGroszkiewicz, 2018-01-02:
This appears to be a side effect of my recent upgrade to version 1.2.0. This did not happen on my Windows machine, where the filters are set at -95 to +95.

RickGroszkiewicz, 2018-01-02:
I actually have the correct filter settings on my Linux laptop.
[image]

MAKOMO, 2018-01-02:
..and no RoR if that filter is deactivated either? I reopen this.

RickGroszkiewicz, 2018-01-03:
Now I am totally confused. Marko sent me an interim version 1.2.0 to solve the issue with "crash happens after deleting event". After installing that version, the background ROR displays properly, and does not disappear after I press Start.
[image]

closed 2018-01-04 by RickGroszkiewicz