artisan-roaster-scope/artisan #148: Phidget ports interferences

benoitmathieu, 2018-01-01:
Hi Marko,

I am running the following Phidgets config : HUB0000 + OUT1100 + TMP1101
K-type TC are attached to TMP1101 ports 0 (BT) and 1 (ET)
FAN is attached to OUT1100 port 1 and Heater driver on port 0
Strangely, FAN and heater sliders values are paired with measured ET value.
If I manually put a higher value on the slider, they keep it; but if I set a lower value (i.e. 0), it always set back to ET measured temp after a few seconds. 
That's a safety problem in my case as it fire the heater back on when I do not expect it.

Thanks a lot for your support on that case.

Here are the screenshots of the observation and config of the sliders.
Slider are configured so they allow a stretching of the working zone of each device.

[image]

[image]

benoitmathieu, 2018-01-01:
here are my artisan settings : 
[artisan-settings.zip](https://github.com/artisan-roaster-scope/artisan/files/1596399/artisan-settings.zip)

MAKOMO, 2018-01-01:
This is because you asked Artisan to do so by defining corresponding event quantifiers. Just untick those quantifiers and you should be good to go.

[image]

closed 2018-01-01 by MAKOMO