artisan-roaster-scope/artisan #149: Save / restore artisan settings - only about 25% correct

RickGroszkiewicz, 2018-01-02:
As part of diagnosing other problems, I saved my artisan settings via Help / Save settings (for later reconstruction), and did a factory reset. After I reloaded my settings file, I noticed that most of my changes to the artisan settings had been lost. I'm not sure if this result is by design, but this is not what I expected.

RickGroszkiewicz, 2018-01-02:
Luckily I had kept careful notes of how I set up artisan on my Linux laptop:
http://www.softwarepolish.com/personal/Artisan%20Linux%20Mint/album/index.html

Most of these settings had to be updated: 
Slide 8 - Config / Device / ET/BT 
Slide 9 - Tools / Extras / Graph
Slide 10 - Config / Autosave
Slide 11 - Config / Events / Config
Slide 12 - Config / Events / Sliders
Slide 13 - Config / Events / Style
Slide 14 - Config / Device / Symb ET/BT
Slide 15 - Config / Sampling interval

MAKOMO, 2018-01-02:
No, this is not by design and was not reported by anybody over the last years versions. Sounds very strange to me as I could load your settings file that you attached to your other issue correctly on my Mac and all those settings were restored. Any messages in the error log on reading in this setting file?

RickGroszkiewicz, 2018-01-02:
Help / Errors shows "Number of errors found 0"

MAKOMO, 2018-01-09:
I took a second look at this. Your settings is totally valid, is loaded correctly on Linux (and Mac OS X) and all settings as stated in the file are reflected after loading. No error is thrown on loading as you also report. So the issue is not in the loading. However, the settings in the file do also not correspond to your settings from the screenshot nor the default settings. This makes me wonder if the settings file you loaded is really the one that you stored before. Difficult to see what went wrong here. Sorry.