artisan-roaster-scope/artisan #150: Call Program in Alarms not working in 1.2

georgejk, 2018-01-07:
Call Program in 1.2 does not call the .bat file having SayStatic text.  It was working with 9.7
[image]

MAKOMO, 2018-01-08:
Works for me under Windows. Just tested with a script `Load Beans.bat` placed on my desktop (path `c:\User\luther\Desktop\Load Beans.bat`) next to the SayStatic.exe with just the following line as content

```
c:\User\luther\Desktop\SayStatic.exe Set heat to 95 percent. Set Fan to 2
```
In the Description cell of the alarm I wrote

```
c:\Users\luther\Desktop\Load Beans.bat
```

Try to specify the full path to your *.bat scripts!

closed 2018-01-08 by MAKOMO

georgejk, 2018-01-08:
Thank you!Geo-----Original Message-----
From: Marko Luther 
Sent: Jan 8, 2018 8:35 AM
To: artisan-roaster-scope/artisan 
Cc: georgejk , Author 
Subject: Re: [artisan-roaster-scope/artisan] Call Program in Alarms not working in 1.2 (#150)

Works for me under Windows. Just tested with a script Load Beans.bat placed on my desktop (path c:\User\luther\Desktop\Load Beans.bat) next to the SayStatic.exe with just the following line as content
c:\User\luther\Desktop\SayStatic.exe Set heat to 95 percent. Set Fan to 2

In the Description cell of the alarm I wrote
c:\Users\luther\Desktop\Load Beans.bat

Try to specify the full path to your *.bat scripts!