artisan-roaster-scope/artisan #153: RoR curve incorrect during roast

Karbug, 2018-02-03:
Since upgrade from Artisan1.1 to Artisan1.2 on HP pc Win8.1, RoR curve begins plotting at roughly the 2:20 mark and seems to proceed normally during the roast. However, once the roast concludes and "End" button is pressed, the RoR curve jumps to the left by about one minute on the y axis so that the curve starts at 1:20. Attached are two captures of the same roast, one taken during the roast showing RoR curve starting at 2:20, the other taken post roast showing curve starting at 1:20. I believe the post roast curve shows actual correct readings but readings during the roast are incorrect. This never occurred with 1.1.
[image]

[image]

MAKOMO, 2018-02-03:
Dear Karburg,

> Since upgrade from Artisan1.1 to Artisan1.2 on HP pc Win8.1, RoR curve begins plotting at roughly the 2:20 mark and seems to proceed normally during the roast. However, once the roast concludes and [...]

What you describe is the expected behaviour and the time-shift is caused by the online smoothing algorithm. Reduce your smoothing settings and the shift-error will reduce. This time-shift is not avoidable in the online-case, basically because the future is largely unknown (cf. https://artisan-roasterscope.blogspot.de/2014/01/sampling-interval-smoothing-and-rate-of.html). However, Artisan applies an optimal smoothing algorithm based on folding in the off-line case that does not produce any time-shift. The use of two different smoothing algorithms did not change between v1.1 and v1.2. What did change is that the not-optimal smoothing algorithm is now also applied to the background profile during recording to at least be able to compare the background/foreground RoR curves directly as they will be both shifted by the same amount. See Section "Rate-of-Rise alignment" of the v1.2 announcement on the Artisan blog: https://artisan-roasterscope.blogspot.de/2017/12/artisan-v12.html.

M

Karbug, 2018-02-05:
Thank you Marko, issue is resolved.

⁣Sent from BlueMail ​

closed 2018-02-08 by MAKOMO