artisan-roaster-scope/artisan #154: Profile issue

MKRoaster, 2018-02-12:
Hi, 

I'm working with artisan and i'm using background profile with Playback Events (by BT).
I still have issuer, that the power change very different - when background has 100% power, artisan suddenly change power to 0%, and after a while change to 100%.
Should we enable or disable any option?
[image]

MAKOMO, 2018-02-12:
Could you please attach here both profiles of this example, foreground and background as well as your settings file (save via menu Help >> Save Settings) for us having a look? Thanks!

MKRoaster, 2018-02-12:
Yes

[Artisan.zip](https://github.com/artisan-roaster-scope/artisan/files/1716462/Artisan.zip)

MAKOMO, 2018-02-12:
Could you please also share your Artisan settings file? That zip only contains the two profiles, but the settings would reflect your full setup to make it possible to trace down your issue. Thanks!

MKRoaster, 2018-02-12:
Yes - in attachement :) Thanks
[artisan-settings.zip](https://github.com/artisan-roaster-scope/artisan/files/1716756/artisan-settings.zip)

MKRoaster, 2018-02-12:
I've made another tests and every time power falls to 0 and i have to put it up manually. I have no idea what is all about.

MKRoaster, 2018-02-12:
[T2.zip](https://github.com/artisan-roaster-scope/artisan/files/1717280/T2.zip)

MKRoaster, 2018-02-13:
Hi, do you have any idea?

MAKOMO, 2018-02-20:
Dear MKRoaster, could you also attach the background profile "Santos T1 18-02-12_1723.alog" used on roasting T2.zip ? Are you sure, that you don't touch the sliders nor the keyboard up/down arrows (which operate the sliders too!) during those roasts? I will check now your roasts on my simulator, but those extra 0 events seem to be random...

MAKOMO, 2018-02-20:
Your "Peru po T6  po BT prawie idealnie.alog" profile from the Artisan.zip has "Peru T6.alog" use the background profile "Peru T6.alog". Could you share this too? Same for the profile "Santos po Test 9 po BT wyà•czyào si© POWER samo.alog" from Artisan.zip. That one is based on "Santos Test 9.alog". Please share those background profile, otherwise it will be hard for me to investigate!

MKRoaster, 2018-02-20:
Hi, i'm sending all files :)

[filesartisan.zip](https://github.com/artisan-roaster-scope/artisan/files/1739952/filesartisan.zip)

MKRoaster, 2018-02-20:
I didn't touch sliders - artisan made changes to 0 by it's own and i't happening every day.

MAKOMO, 2018-02-20:
Hm. All those profiles link to background profiles not contained:

- "Peru po T6  po BT prawie idealnie.alog" links to the background profile 'C:/Users/Joanna/Desktop/ARTISAN PROFILE/Peru T6.alog'
- "Santos po Test 9 po BT wyà•czyào si© POWER samo.alog" links to 'C:/Users/Joanna/Desktop/ARTISAN PROFILE/Santos Test 9.alog'
- "Santos T1 18-02-12_1723.alog" links to 'C:/Users/Joanna/Desktop/ARTISAN PROFILE/Kolumbia T1.alog'
- "T2.alog" links to the background profile 'C:/Users/Joanna/Desktop/ARTISAN PROFILE/Piotrek/Santos T1 18-02-12_1723.alog'

Could you please attach the files

- 'C:/Users/Joanna/Desktop/ARTISAN PROFILE/Peru T6.alog'
- 'C:/Users/Joanna/Desktop/ARTISAN PROFILE/Santos Test 9.alog'
- 'C:/Users/Joanna/Desktop/ARTISAN PROFILE/Kolumbia T1.alog'
- 'C:/Users/Joanna/Desktop/ARTISAN PROFILE/Piotrek/Santos T1 18-02-12_1723.alog'

to allow me to reproduce this effect!

Thanks,
Marko

MKRoaster, 2018-02-21:
I've attached the files, Thanks, Peter

[profiles.zip](https://github.com/artisan-roaster-scope/artisan/files/1745468/profiles.zip)

MAKOMO, 2018-02-21:
Thanks! Very helpful. I got an idea of what is going wrong. I'll try to fix this...

MKRoaster, 2018-03-05:
Hi Marko, did you find, why our artisan changes power different than we have set in background?

MAKOMO, 2018-03-12:
Dear MKRoaster,

I just fixed that issue. Replay-by-temperature events could be executed out-of-order. As in all those cases you had a later event moving the slider 0 at about the same temperature, this event was triggered in the middle of the roast (out of order). In case you want to have a pre-release of the next Artisan version with this issue fixed to test, send me a message via the Artisan blog contact formular at https://artisan-roasterscope.blogspot.de/p/contact-me.html indicating your OS (Mac or Windows).

Thanks for reporting this one,
Marko

closed 2018-03-12 by MAKOMO