artisan-roaster-scope/artisan #155: Phases Dialog Box Behavior Mac vs PC

stinsonddog, 2018-02-16:
On my MAC if I select Config>Phases, and I check Auto Adjusted, all of the temp settings between Drying Min and Finishing Max are greyed out.  This doesn't happen on a PC.  I had an issue where my Drying Phase Green Bar didn't line up with my DE Watermark.  This was on the MAC and it went away if I chose Auto Adjusted and not Auto Dry.  But when I did that these temps got greyed out.  It only happens on the Filter tab and not the Espresso Tab leading me to believe its a bug.

stinsonddog, 2018-02-16:
It's not clear to me what the relationship between Statistics and Phases is when one selects Auto Adjusted vs Auto Dry.  It's when Auto Adjusted isn't selected that the Statistics green bar doesn't line up with the DE watermark.  Should it?  What does it mean if it doesn't - ie Auto Dry is selected?

MAKOMO, 2018-02-20:
Thanks Michael and Terry for raising this issue. There was a confusion on enabling/disabling those widgets in the Espresso tab. Fixed in source. Will be part of the next release.

Regarding your question: in AutoAdjusted mode, the phases limits are taken from the events of the profile. In manual mode its somehow the other way around. You define the limits for the phases (those might be targets, defaults or common sense knowledge). In this case of course the events like DRY and FCs might not correspond to the phases limits. Finally, in this manual mode you can instruct Artisan to automatically key the DRY and FCs events for you once the BT reaches the (set) limit of the corresponding phase. Again (as Artisan is not looking at your beans during roasting), the beans might not yet DRY once the limit is reached.

If you ask now, why all these modes/possibilities are available: because the users asked for them. It seems that it is quite hard to find two roast masters with the same concept...

closed 2018-02-20 by MAKOMO