artisan-roaster-scope/artisan #158: Lowercase delta doesn't render

rpaulo, 2018-03-08:
Commit 104aab741a5450e0b362d00dbd90daa16541797b introduced the lower case greek delta letter.  This doesn't render correctly on my system.

I'm on Debian buster.

MAKOMO, 2018-03-08:
Thanks for the hint, I'll take a look. In return, could you take a look at the Aillio profile importer and align it with your work on reading the Aillio? I think that importer also names/adds some extra devices and one should take care that order and names align with your default machine setup. Maybe everything is ok already, but I can not really test this.

rpaulo, 2018-03-08:
Attached is a screenshot of an R1 profile imported into Artisan.  As
you can see, the profile just shows ET, BT and RoR, so everything looks
fine.

You can see the lower case delta display problem on the graph.  On the
LCD display, you can see a correctly rendered lower case delta, but
that's because I fixed the code.

MAKOMO, 2018-03-08:
Cannot see that LCD. What is your fix? Should we got the full way and rename those DeltaET/DeltaBT to the by now more standard, thus longer, names "ET RoR" and "BT RoR", avoiding the Greek letters?

rpaulo, 2018-03-08:
Oops, I thought the LCD was included.  I don't know how to get the greek letter on the combo boxes but if you use "&delta;" on the LCDs, it works.
I actually like ΔBT.  I'm not sure what was behind the change to lower case delta, but I think the upper case delta is more correct.
It's true that a lot of people talk about RoR but that's an English acronym.  I wonder if RoR became more popular because people don't remember the meaning of the Greek letter Delta?

MAKOMO, 2018-03-08:
Hm. I like the Greek large delta too, just switched to the small one as on some fonts the large one didn't show enough spacing.  Still you are right, the \Delta is more correct than the RoR, as the RoR works only for "rising" curves, right;). So let's try to go for big \Delta. 

In v1.2 we used this HTML variant

`self.label5.setText("<big><b>&Delta;" + u(QApplication.translate("Label", "BT",None)) + "</b></big>")`

on the LCD labels. Does this work on your platform? I would prefer this one to adding a non-ascii character in the source.

The problem is that on some fonts used in the graph, like the included Humor, there is not Greek symbol at all, in the case of Humor not even a small letter d. Hm. My goal is to make this somehow consistent and understandable and breve and to make it work on all platform and occurrences. What is your suggestion?

MAKOMO, 2018-03-08:
Easiest would be to use just "dET" and "dBT". Not as nice, but that would work across the app and all fonts. Just for the Humor, lacking the small d it would be substituted automatically to a capital D, like "DET" and "DBT".

rpaulo, 2018-03-08:
Yes, the big delta in HTML format works fine, what didn't work was unicode delta.
Since the font ships with Artisan, can't you add the Delta symbol to it?

MAKOMO, 2018-03-08:
Ok. Found a way to use the large Greek Delta in LCD labels (Qt rendering) using the HTML entity "&Delta;" as above and a math font symbol on the graph (matplotlib rendering) using $\Delta$ that even renders correctly if the Humor font is selected as for that character a different font is uses, independent from the Artisan font setup. If that would work also on your platform, this might be the way forward. I might try to render that large Delta some points smaller than the basic text font thus... I will see what looks better.

[image]

rpaulo, 2018-03-08:
The Delta is a little bit off in the graph but it's a nice solution.

MAKOMO, 2018-03-08:
Yep, this is the Humor font and is part of its craziness. I tried to make the \Delta a little smaller (works only on Qt), but that does not look good. However, adding a small space between the Greek symbol and the ET/BT looks good. I try to go with this. I plan to go with the "\Delta BT" variant shown in this screenshot. Ok?

[image]

rpaulo, 2018-03-08:
Looks good to me.  What about the check boxes in Tools > Extras... ?

MAKOMO, 2018-03-08:
Yes, I'll change those too. Try to make this consistent across the whole app and let you then check;)

Regarding the Aillio import: in the code I see a

`# add extra device if exitTemperatures`

So this extra temperature curve is named "exitTemperature". This is not consistent with your work, right? Could you please check again, while I fixe the deltas...

rpaulo, 2018-03-09:
I don't think Aillio's application tracks exitTemperature.  The JSON reads:
```
"exitTemperature":[]
```

rpaulo, 2018-03-09:
I think that we should probably not even plot that until Aillio starts using like.  Like I said before, I'm not sure what this temperature is tracking since it sits at -60C almost all the time

MAKOMO, 2018-03-09:
I close this as the issue with the Greek Delta symbol is resolved by now.

closed 2018-03-09 by MAKOMO