artisan-roaster-scope/artisan #159: UI style is no longer saved

rpaulo, 2018-03-09:
Commit 220ca099b76bd16519fafc28d6250f2cc12e5513 broke the UI style.  If I chose "Cleanlooks" and then exit Artisan, the style is not restored.

rpaulo, 2018-03-09:
@roasterdave you committed this code.  Any ideas what might be wrong?

MAKOMO, 2018-03-09:
Rui, I cannot duplicate this, but I also have some strange issues on settingsLoad. I guess there is an uncatched exception happening on settings load, but just in some cases.  Cloud you trace this down to where this happens with the current trunk?

I added and removed some settings in my latest commits and might have broken something, I just can't catch it...

MAKOMO, 2018-03-09:
Now I understood what you reported. The appearance item in settings files was just plain empty. I now fixed this and the issue seems to be caused by a change in the underlying Qt/PyQt stack that is not reporting the current style in use anymore. Might be a bug on their side or just an API change. I resolved that by introducing our own internal variable to track the style the user selected last.

closed 2018-03-09 by MAKOMO

tanius, 2019-04-30:
Is it still possible to choose the UI style in current versions of Artisan?

I just can't find it (Artisan 1.6.2 here).

MAKOMO, 2019-05-03:
menu Config >> Curves, last tab UI