artisan-roaster-scope/artisan #163: Config >> Curves menu opens the Extras dialog

rpaulo, 2018-03-19:
We need to fix the Config >> Curves menu because it opens the Extras dialog.
The extras dialog is also no longer available in the help menu.

MAKOMO, 2018-03-19:
I'll take care of this today as I created that mess yesterday (without much testing).
M

MAKOMO, 2018-03-19:
So I just renamed the Curves dialog. This all is the former Extras dialog. As it mostly contains curve configurations I moved it under the Config menu based on a suggestion from Michael. This is not fully ok, as HUD, Plotter and Math are more "Tools", but mostly curve related. The UI tab contains some more general settings, but also the WebLCDs configuration. In the end one would need to split this up in several parts. I leave this for v1.4

closed 2018-03-19 by MAKOMO