artisan-roaster-scope/artisan #164: Clean up the src directory

rpaulo, 2018-03-24:
Lots of files spread around in the src directory.  We could do some cleanup.

Suggestion:
Move Comm* to a separate directory.
Move icons to the icons directory.
Move build/setup scripts to their own directory.

MAKOMO, 2018-03-24:
Yep, please go. Also the build scripts could be cleaned up, eg. the build-mac35.sh script could be renamed into build-mac3.sh as it covers both v3.5 and v3.6 now.

rpaulo, 2018-03-24:
@MAKOMO Can you delete the scripts that are simply not used anymore?  I know the scripts that Travis/AppVeyor uses, but I'm not sure which ones you might use...

MAKOMO, 2018-03-24:
Yep. One moment I have to check.

MAKOMO, 2018-03-24:
Done. Did not rename the build-mac35 script not to break anything in your scripts

closed 2018-04-18 by rpaulo