artisan-roaster-scope/artisan #167: Changed Computer, Now have downward spikes while roasting

curtismartin252, 2018-03-25:
I went from a mac mini, running just fine with buttery smoothe graphs. To new (latest model) MacBook Pro with downward spikes while roasting.

I install artisan, copy over my settings file, download and install the phidget 1048 driver. All looked good unstill I actually started roasting. Then I would get spikes in my graph dropping down and rising back up. I never got a spike to rise up above profile, just down.

I am trying to figure out what changed and how to fix it. The roaster (Huky 500T) and phidget never changed. If I hook them back up to my mac mini all is fine. I believe the issue has to do with my MacBook Pro being a laptop and maybe its a battery or AC power related issue. Just can't seem to nail anything down. It happens when running off of the battery, or when I plug it in.

[image]
[image]

MAKOMO, 2018-03-26:
Could you please attach the corresponding .alog file here?

curtismartin252, 2018-03-26:
Yes, here they are.

[Archive.zip](https://github.com/artisan-roaster-scope/artisan/files/1847885/Archive.zip)

MAKOMO, 2018-03-26:
Thanks for forwarding the profiles. I took a look and have no real good news for you. It seems to me that you have an issue with your BT probe (ET and communication seems to be fine). The internal insolation might be broken. What seems to happen that by the movement of the beans a short circuit happens sometimes, which results in low (zero) readings within the Phidget. As the Phidget is averaging some values the resulting values are not really zero, but in between the real values. Artisan is receiving those wrong values and not (communication) errors and thus just logs those. If the issue would be a communication error, Artisan would log a -1 and the new drop-out prevent algorithm in v1.2 would just interpolate the missing values such that you would not see the mess. Just in your unlucky situation this mechanism cannot be triggered as it does not know that the values are invalid. 

First thought there is an issue with this mechanism in case oversampling is active, but I just tested with your profile and that works fine (you could test your self by deactivating oversampling). All you can do for now, ensure that the mechanism is triggered by activating an output filter and allow only temperatures in the range of 100-500F. That would render those lower values as invalid (-1) and thus trigger the approximation algorithm. You will not see those dropouts anymore (but they are still there).

So this is not a noise or communication issue. Your signals look fine and almost noise free.

I'll close this issue as this seems to me not an issue within the Artisan software.

closed 2018-03-26 by MAKOMO

curtismartin252, 2018-03-31:
So would you recommend a new Phidget 1048 and or thermocouple? Would that
make the problem go away?

MAKOMO, 2018-04-01:
The Phidget seems to be fine, but I would test this one thermocouple. Maybe you can swap it with the other and see if the issue goes away (as the ET probe is not in contact with anything a small defect on its insulation might not have any consequences.

M

jasonwardenburg, 2018-04-03:
I had that same issue and upgrading to the new version of artisan resolved the issue for me. I was looking at my amprobe temps, and they seemed to be fine, the issue was limited to the MET curve in artisan (had the same result when I switched TC inputs). 

after upgrade issue was resolved.