artisan-roaster-scope/artisan #171: Index out of bounds

rpaulo, 2018-04-01:
Got a report from a user saying Artisan is still crashing on updateLCDtime.

[image]

```
            if self.timeindex[0] != -1:
                ts = tx - self.timex[self.timeindex[0]]
```

This means that self.timeindex[0] is out of range.

MAKOMO, 2018-04-01:
Bad! We cannot fix this here as timeindex is an array of fixed size of length 8 that is initialized at startup. This array changes its values but never its size. We have to find out where this array gets messed up. This user connecting to an Aillio? We should first look in this part of the code if something can go wrong there.

rpaulo, 2018-04-01:
I don't know how the Aillio code could mess up the timex array.  This user has a lot of events.

MAKOMO, 2018-04-01:
Can we get our hands on his exact setup and a simple procedure on how to reproduce this or at least some hint when this happens?

rpaulo, 2018-04-01:
This happens at the end of the roasting session when the R1 changes state, which will trigger an event.  Attached are the events.

[image]

MAKOMO, 2018-04-01:
I see timeindex being set (besides the initialization to a constant array of 8 items) being set in

- loadpoints(): loading points generated and stored to a file before using the designer
- setProfile(): loading a profile (however, reset() calls clearMeasurements() which initialize the array again correctly before being able to record anything.
- importBullet(): ??

So is this crash happening during recording? Does he use the designer? Did he import a profile using the designer?

MAKOMO, 2018-04-01:
What exactly do you mean by triggering an event? An alarm that triggers a button action? I am very confused now. Can we now which button is triggered by which alarm, then we could test this a bit?

rpaulo, 2018-04-01:
I don't have the configuration yet.

MAKOMO, 2018-04-01:
Could it be a parallel processing issue? The easiest fix would be just to catch this in the timer update function and just not update the timer if something goes wrong, but I would love to understand exactly how this array can get in such a bad state as a lot other functions depend on its size of 8 and might crash as well.

raescuin, 2018-04-01:
Hello Makomo, I've been testing Artisan for Aillio Bullet. After download this morning, I tried roasting some beans and when I aborted through the Bullet PRS, that window above pop up. I tried again and in the early part of roasting it popped up a few times again but the program did not crash, which is good I did not have to abort the roast.. The roast continued until the end.
config file sent to Rui. Let me know if there's anything you need.

thanks Robert

raescuin, 2018-04-01:
here it is. settings
[artisan-settings 4012018-149pm.zip](https://github.com/artisan-roaster-scope/artisan/files/1866726/artisan-settings.4012018-149pm.zip)

raescuin, 2018-04-01:
fyi, I had a saved designer profile in the background, if that will help troubleshoot..

rpaulo, 2018-04-02:
Does the crash reproduce without the background profile?

raescuin, 2018-04-02:
I did not try without the background, I'll try to reproduce today w/o bg

MAKOMO, 2018-04-02:
Could you tell us how you produced exactly that background and maybe send us the corresponding .alog file? Thanks!

raescuin, 2018-04-02:
made a roast profile through designer, here's the .alog
[Espresso Profile v1.1.zip](https://github.com/artisan-roaster-scope/artisan/files/1867924/Espresso.Profile.v1.1.zip)

raescuin, 2018-04-03:
I tried duplicating this error yesterday but it did not happen. I will keep you updated if it happens again. Thanks

rpaulo, 2018-04-03:
When it did not reproduce, did you have a background profile?

raescuin, 2018-04-03:
I tried both but nothing. That day I had problems with my alarms keep
popping, I fixed my alarms yesterday before trying.

rpaulo, 2018-04-08:
Should be fixed by bb02be42d9c949edba0f325eb3e74da895f37da0

closed 2018-04-08 by rpaulo