artisan-roaster-scope/artisan #173: Charge Button

raescuin, 2018-04-06:
remains enabled after charge. 1.3.0
[image]

raescuin, 2018-04-06:
after changing my graph colors, when I click on Charge, the button disappears but the "Charge" text I can see, when I click on the text, it stops the charge. attached
[RAE Theme 04042018-606am - Copy.zip](https://github.com/artisan-roaster-scope/artisan/files/1883786/RAE.Theme.04042018-606am.-.Copy.zip)

MAKOMO, 2018-04-06:
Thanks for sending the theme. What you experience is the intended behaviour. On pressing any of those buttons once it marks the event and "flattens" the button. However, the button stays there (in your case you only see its label). Pressing it again rewokes the mark and "unflattens" the button again. This is a way to undo wrongly placed event marks.

closed 2018-04-06 by MAKOMO