artisan-roaster-scope/artisan #176: Printing doesn't work on Linux

rpaulo, 2018-04-12:
I cannot print a roast profile on Linux.   It works fine if I print to a file, but it doesn't seem to detect my printer correctly.
If I run it from source, it works fine, so it could be something related to how Artisan is packaged.

MAKOMO, 2018-04-12:
Rui

could you check if the Linux variant of the

QtPrintSupportFramework (a shared lib)

and the

qt_plugins/printsupport/libcocoaprintersupport.dylib

must also be a shared lib is contained in the Linux builds?

That's the only thing that rings on this in my head.

M

rpaulo, 2018-04-13:
I have

```
./libQt5PrintSupport.so.5
./PyQt5/Qt/plugins/printsupport
./PyQt5/Qt/plugins/printsupport/libcupsprintersupport.so
./PyQt5.QtPrintSupport.so

```

MAKOMO, 2022-02-06:
fixed by moving up to Qt6/PyQt6

closed 2022-02-06 by MAKOMO