artisan-roaster-scope/artisan #178: matplotlib not using persistent font cache

mreppen, 2018-04-16:
## Expected Behavior
Use the default matplotlib font cache location, e.g., `~/.cache/matplotlib` or artisan-specific directory like `~/.cache/artisan`.

## Actual Behavior
Uses temporary location, and thus has to rebuild cache at every run (takes a long time on my system).
In my case, it is written to a subdirectory of `/tmp/`, which of course is different every time artisan is run.

## Specifications
  - Artisan Version: 1.3.0
  - Platform (Mac/Windows/Linux + OS version): Arch Linux running from .deb file (same on rpm)

Although I'm not using a supported distribution, the fact that the cache is stored in a temporary directory suggests the problem would occur also on other distributions.

MAKOMO, 2018-04-16:
Seems to be a platform specific issue. On Mac OS X (build with py2app)  the users standard .matplotlib font cache is used or created/updated. Could be an issue on pyinstaller builds (Linux/Windows). Found this one: https://github.com/pyinstaller/pyinstaller/issues/617

mreppen, 2018-04-18:
The description of the behavior is not entirely clear to me, likely because I don't really know PyInstaller.

It seems to me like the hook with that comment is responsible for setting up the new temporary mpl directory. Why couldn't it just copy/link the `fontList.cache` from the default directory?

closed 2018-04-19 by rpaulo

rpaulo, 2018-04-19:
https://github.com/pyinstaller/pyinstaller/issues/3472

commit daec49e5 (2018-04-19): Render pyi_rth_mplconfig.py useless because it makes our startup slow on Linux

Closes #178.