artisan-roaster-scope/artisan #184: No communication between Artisan(MAC) and Aillio Bullet

thomashaugaard75, 2018-04-22:
## Expected Behavior
Registry of temperature and control of Aillio Bullet Roaster.

## Actual Behavior
No communication with Artisan. Error message when turning Artisan on = "Aillio R1: Not found or no permission. 

## Steps to Reproduce the Problem

  1. Turn on Artisan and Aillio R1
  2. Press "On" in Artisan
  3.

## Specifications

  - Artisan Version: Latest
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version): Latest Sierra
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

rpaulo, 2018-04-22:
@thomashaugaard75 can you run ioreg -l on the command line?

thomashaugaard75, 2018-04-22:
Sorry. Only saw your message now and I have gone to bed. Will do it right away tomorrow. 

Sendt fra min iPhone

thomashaugaard75, 2018-04-23:
This is what I got.

Last login: Mon Apr 23 07:45:32 on ttys000

Fridas-MBP:~ fridahaugaard$ ioreg -i

+-o Root  <class IORegistryEntry, id 0x100000100, retain 18>

  +-o MacBookPro9,2  <class
IORegistryEntry:IOService:IOPlatformExpertDevice, i$

    +-o AppleACPIPlatformExpert  <class
IORegistryEntry:IOService:IOPlatformExp$

    | +-o IOPMrootDomain  <class IORegistryEntry:IOService:IOPMrootDomain,
id 0$

    | | +-o IORootParent  <class IORegistryEntry:IOService:IORootParent, id
0x1$

    | | +-o RootDomainUserClient  <class
IORegistryEntry:IOService:IOUserClient$

    | | +-o RootDomainUserClient  <class
IORegistryEntry:IOService:IOUserClient$

    | | +-o RootDomainUserClient  <class
IORegistryEntry:IOService:IOUserClient$

    | | +-o RootDomainUserClient  <class
IORegistryEntry:IOService:IOUserClient$

    | | +-o RootDomainUserClient  <class
IORegistryEntry:IOService:IOUserClient$

    | | +-o RootDomainUserClient  <class
IORegistryEntry:IOService:IOUserClient$

    | | +-o RootDomainUserClient  <class
IORegistryEntry:IOService:IOUserClient$

    | | +-o RootDomainUserClient  <class
IORegistryEntry:IOService:IOUserClient$

    | | +-o RootDomainUserClient  <class
IORegistryEntry:IOService:IOUserClient$

    | | +-o RootDomainUserClient  <class
IORegistryEntry:IOService:IOUserClient$

    | | +-o RootDomainUserClient  <class
IORegistryEntry:IOService:IOUserClient$

    | | +-o RootDomainUserClient  <class
IORegistryEntry:IOService:IOUserClient$

    | | +-o RootDomainUserClient  <class
IORegistryEntry:IOService:IOUserClient$

    | | +-o RootDomainUserClient  <class
IORegistryEntry:IOService:IOUserClient$

    | | +-o RootDomainUserClient  <class
IORegistryEntry:IOService:IOUserClient$

    | | +-o RootDomainUserClient  <class
IORegistryEntry:IOService:IOUserClient$

    | | +-o RootDomainUserClient  <class
IORegistryEntry:IOService:IOUserClient$
[... 2001 lines cut ...]

      | +-o IOSurfaceRootUserClient  <class
IORegistryEntry:IOService:IOUserCli$

      | +-o IOSurfaceRootUserClient  <class
IORegistryEntry:IOService:IOUserCli$

      | +-o IOSurfaceRootUserClient  <class
IORegistryEntry:IOService:IOUserCli$

      | +-o IOSurfaceRootUserClient  <class
IORegistryEntry:IOService:IOUserCli$

      | +-o IOSurfaceRootUserClient  <class
IORegistryEntry:IOService:IOUserCli$

      +-o IOUserEthernetResource  <class
IORegistryEntry:IOService:IOUserEthern$

      +-o com_apple_driver_pmtelemetry  <class
IORegistryEntry:IOService:com_ap$

Fridas-MBP:~ fridahaugaard$

2018-04-22 23:51 GMT+02:00 Thomas Haugaard <thomas.th.haugaard@gmail.com>:

rpaulo, 2018-04-23:
Is your account administrator?
https://support.apple.com/kb/PH25796?locale=en_US

thomashaugaard75, 2018-04-23:
I am completely new to this. I do not know if my account is administrator. 

Sendt fra min iPhone

rpaulo, 2018-04-23:
Go to system preferences and then to users.  If your user is an administrator, it should say so.

thomashaugaard75, 2018-04-23:
So you are asking if I am an admin on my Mac. If that is the case then yes. It is my private Mac. 

Sendt fra min iPhone

rpaulo, 2018-04-23:
Can you send me the output of:

1)  ioreg -p IOUSB -w0 -l
2) system_profiler SPUSBDataType

thomashaugaard75, 2018-04-23:
> Can you send me the output of:
>
> ioreg -p IOUSB -w0 -l
> system_profiler SPUSBDataType

I will send tomorrow. I am on my way to Finland. 

Best 
Thomas

Sendt fra min iPhone

labpartner, 2018-04-23:
I am having the same issue, but I'm on win10. I just received my bullet two days ago and I suspect the OP is having the same problem.

The new units have a "rev3" board. They didn't even work with RoastTime 1.0.0.49 (not detected) and they had to release a 1.0.0.50 solely to address this problem. I suspect this is the reason why Artisan isn't recognizing either.

rpaulo, 2018-04-24:
@labpartner can you find out what’s the VID/PID/Interface/etc.?

labpartner, 2018-04-24:
USB\VID_0483&PID_A27E&MI_01\6&3820E00A&1&0001

labpartner, 2018-04-24:
that fixed my issue, thanks! Hopefully that was it for OP. Excited to use the software soon.

rpaulo, 2018-04-24:
Great, please let me know if everything works.

rpaulo, 2018-04-24:
@thomashaugaard75 please try this build https://dl.bintray.com/artisan/artisan-artifacts/:artisan-mac-1.3.1.dmg

labpartner, 2018-04-24:
I'm playing around with everything (did not do a roast). I notice an unrelated issue with the sliders: they do not seem to set to the pre-programmed values onces "roasting" has started. eg. power shows "10" despite being 90. should I open a new issue? Also I notice that drum speed isn't adjustable

rpaulo, 2018-04-24:
1) Unplug the roaster from power
2) Plug it back in and wait for it to be ready
3) Plug the USB cable and start Artisan
4) Press ON.
5) Put the roaster in roasting mode.
6) Interact with the roaster using the control panel or the sliders.

Are they not working?  If they aren't then the protocol changed.

labpartner, 2018-04-24:
ok, sorry you are correct, I must have had a different order of operations (roaster plugged in prior to starting Artisan).

rpaulo, 2018-04-24:
The problem is that the roaster keeps all the variables in memory and when you connect using Artisan or Roasttime it will stream them.  You can see this in Roasttime when it suddenly plots the graph very fast.
Artisan doesn't support this method, so we work around by starting Artisan before you start pre-heating.

raescuin, 2018-04-24:
fyi, just tried v1.3.1 on my older Bullet and no issues.

thomashaugaard75, 2018-04-24:
Just came home. The link says it does not exist.

rpaulo, 2018-04-24:
https://dl.bintray.com/artisan/artisan-artifacts/artisan-mac-1.3.1.dmg

thomashaugaard75, 2018-04-24:
I get this error

*Artisan” kan inte öppnas eftersom det kommer från en oidentifierad
utvecklare.*

*Which means*

*Artisan can not open because it comes from an unidentified developer.*

*Thomas*

thomashaugaard75, 2018-04-24:
Sorry for simple inconvenience. I can run the prog now. Will test roasting tomorrow. Many thanks!

Sendt fra min iPhone

closed 2018-04-25 by rpaulo

thomashaugaard75, 2018-04-25:
I just ran my first batch since the patch that at least fixed the connection. Several issues seem to still be problematic: 
First of all, there is no RoR curve on the screen despite having selected to show it
Once the roast is over the chart looks very strange (see below picture)

As you see, no RoR and datapoints are gone. I am furthermore unable to save it and reopen it. 

/Thomas 

Sendt fra min iPhone

thomashaugaard75, 2018-04-25:
It now seems I have been able to add the RoR curve after having chosen to configure machine like Bullet R1. However I get a message that artisan is unable to claim interface. 

Sendt fra min iPhone

rpaulo, 2018-04-26:
I cannot see any picture.  Usually the interface claim message means something else on the system, like Roasttime, is claiming the interface and Artisan can't.

thomashaugaard75, 2018-04-26:
All other programs should have been shut down but I will check again. Now the RoR curve is missing again. Very strange. 

/Thomas 

Sendt fra min iPhone

commit 37b5892f (2018-04-24): Match the new Aillio PID.

Possibly fixes #184.