artisan-roaster-scope/artisan #185: Programmable pre-heating

rpaulo, 2018-04-23:
For pre-heating a machine, most roasters have a protocol.  It would great if Artisan could automate this process.
We could have a dialog that can be used to configure a pre-heating protocol and then we Artisan would follow it.

For example, a protocol could be:
1) pre-heat at setting 7 fan 0 to BT 165C
2) switch to setting 3 until BT reads 160 for 10 minutes
3) switch to setting 1 until BT reads 155 for 10 minutes

This may be implemented with alarms, but I'm not sure.

raescuin, 2018-04-23:
Great idea, but would this override the Bullets' default preheat settings? I'll try it next with alarms and report back.

rpaulo, 2018-04-24:
You would have to put the bullet in roast mode so that Artisan can control all the heat and fan. The PH algorithm in the bullet wouldn’t work here.

raescuin, 2018-04-24:
just tested using alarms, I'm able to control the fan but not the heat.

rpaulo, 2018-04-27:
One way to accomplish this is to record a pre-heat profile and replay that using the PID.

mfarre, 2020-06-14:
I find @rpaulo's dialog idea very interesting. I would use it as well.
@MAKOMO @rpaulo if you can provide some guidelines of how to structure such a feature (for example where to put the dialog to keep it consistent, what do you think could be reused from the existing code, etc), I could take a look and do a pull request.

MAKOMO, 2021-08-09:
Should be covered by the various structures added in v2.4.4: see [Supporting Between Batch Protocols](https://artisan-roasterscope.blogspot.com/2020/12/supporting-between-batch-protocols.html)

closed 2021-08-09 by MAKOMO