artisan-roaster-scope/artisan #187: Artisan 1.3 crash on Raspbian

testabean, 2018-04-25:
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1. Open Artisan
  2. make operations 
      Save as 
      Open
      Save Settings as
      
  3. Artisan crash completely

## Specifications

  - Artisan Version: 1.3.0
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version): Raspian Stretch
  - Connected devices or roasting machine: any
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

MAKOMO, 2018-04-25:
Which hardware are you running Stretch on?

MAKOMO, 2018-04-25:
Does "Save" work? Just asking as it is not in your list of crashing commands or is it crashing on all operations opening a file selector?

testabean, 2018-04-25:
Yes, crashing on all operations opening a file selector. It's a new raspberry Pi 3 B+

DavidFW1960, 2018-04-27:
Same. I was running the Jessie build on a Pi 3B and v1.2 worked fine no problems. 1.3 crashes when I edit alarms.
If I just start a roast with my previous alarms and settings it works fine.

bigl00z3, 2018-05-03:
I can confirm the same behaviour in 1.3 (and 1.3.1) on Stretch when:

- saving settings
- changing background graph

Artisan crashes (quit)

Moving back to version 1.2 solves the problem. Using 1.3 on OS X works without issues.

samtsai5212, 2018-05-08:
Artisan 1.3 crashes probably because it tried to save something without permission. Try to run Artisan in terminal with root privilege:
_sudo artisan_

If you want to make it permanent, run the following:
_sudo nano /usr/share/applications/artisan.desktop_
find the line:
_Exec=artisan %U_
Change to:
_Exec=sudo artisan %U_
press ctrl+X then 
press  Y then eneter

MAKOMO, 2018-05-09:
Thanks @samtsai5212 for this valid workaround which seems to work fine. We are investigating what the cause of this is and try to fix it..

commit a334b550 (2018-05-14): Workaround for RPI issue where GTK 3 throws an assertion in the Save dialog.

Fixes #187
closed 2018-05-14 by rpaulo