artisan-roaster-scope/artisan #192: Unhandled Exception (Config/Alarms)

arkenamun, 2018-05-01:
## Expected Behavior
Edit/compare 2 alarm files, 2 instances open simultaneously

## Actual Behavior
[image]
  File "artisanlib\main.py", line 44208, in <lambda>
  File "artisanlib\main.py", line 44347, in addalarm
  File "artisanlib\main.py", line 44714, in markNotEnabledAlarmRows

--------------------------------------------------------------------------------
artisanlib\main.py::<lambda>:44208
	                   _ = False
	                self = <artisanlib.main.AlarmDlg object at 0x0000025205F0D9D8>
artisanlib\main.py::addalarm:44347
	              header = <PyQt5.QtWidgets.QHeaderView object at 0x0000025209DD05E8>
	             nalarms = 76
	                self = <artisanlib.main.AlarmDlg object at 0x0000025205F0D9D8>
artisanlib\main.py::markNotEnabledAlarmRows:44714
	                   j = 0
	                   i = 56
	                self = <artisanlib.main.AlarmDlg object at 0x0000025205F0D9D8>

Error log:
version = 1.3.0

2 22:12:22.628 2018-05-01, 22:12:22 -------------------------------------------------------------------------------- : list index out of range

1 22:11:56.424 Exception: createalarmtable() list index out of range@line 44761

## Steps to Reproduce the Problem
Is not immediately reproducible.

## Specifications

  - Artisan Version: 1.3.0
  - Artisan Build (number in brackets shown in the about box): 95bf230b, (Brad, 

Australia, fixes Pulse not ending)
  - Platform (Mac/Windows/Linux + OS version): W10
  - Connected devices or roasting machine: Modbus, Phidgets and Extech
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

ASET NOT A SUPPORTED FILE TYPE!!

RickGroszkiewicz, 2018-05-01:
You should copy, then rename the file as *.aset.txt to upload it

MAKOMO, 2018-05-07:
Added a handler to catch this one. The original source of the issue remains unclear and cannot be reproduced.

closed 2018-05-07 by MAKOMO