artisan-roaster-scope/artisan #193: Split graphing into two sections: 1)time/temp curves   2)power levels

InvisibleOrdinary, 2018-05-02:
## Expected Behavior

This is really a feature / design request: To be able to see the percentage to set power/fan or other input during roast (whether current roast or background profile) separated (on it's own axis) separate from the temp so that we could more easily read the power level.

## Actual Behavior

power levels are included at the bottom of the time/temp graph, so the time is viewable, but the level you set at is largely a guess since the scale is merged with the temperature axis

## Steps to Reproduce the Problem

  1. Open Artisan
  2. Set gas power level or air fan power level via sliders

## Specifications

  - Artisan Version: 1.3
  - Artisan Build (number in brackets shown in the about box): ca3c5f1
  - Platform (Mac/Windows/Linux + OS version): Mac OS
  - Connected devices or roasting machine: Proster via Phidgets 1048… looking at upgrading to Phidgets VINT to record non-temp data directly
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.

[180503_artisan-settings.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/1968930/180503_artisan-settings.aset.zip)

Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-07-07:
This goes against the current design of Artisan having an integrated display of all data. As such this is unlikely to change in the near future. Still we can keep this request open for now.

InvisibleOrdinary, 2018-07-09:
Hmm, then perhaps there can be some indication of the scale for that lower section? The underlying issue is that currently, there is no way to see what the graph values are there, so when using it as a reference profile, it's only relative up or down, but not 60% 30% or whatever. Seeing the values would make it much more useful.

MAKOMO, 2018-10-20:
You can click on those nodes to get all the details including the event values printed in the Artisan message line. Further you can select a different custom event rendering. In Artisan v1.3 we added "Combo" type of rendering that places the event values directly on the event graph. See the corresponding blog https://artisan-roasterscope.blogspot.com/2018/04/artisan-v13.html and check the 5 event rendering possibility under menu Config >> Events.

closed 2018-10-20 by MAKOMO