artisan-roaster-scope/artisan #194: Mac OS Cmd-S should be Save As

InvisibleOrdinary, 2018-05-02:
## Expected Behavior

Mac OS : standard OS behavior is that Cmd-S is save as… 

## Actual Behavior

Cmd-S toggles slider visibility on/off

## Steps to Reproduce the Problem

  1.  Open Artisan
  2. type Cmd-S
  3. repeat

## Specifications

  - Artisan Version: 1.3
  - Artisan Build (number in brackets shown in the about box): ca3c5f1
  - Platform (Mac/Windows/Linux + OS version):Mac OS 10.13.4
  - Connected devices or roasting machine: phidgets 1048
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.

[180503_artisan-settings.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/1968947/180503_artisan-settings.aset.zip)

Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-05-03:
I just tried, but for me on the very same OS, with your settings loaded, Cmd-s opens the file selector if no file path is set or if a file path exists it saves to that file. Just s views or hides the sliders. Could you please double check and report here?

InvisibleOrdinary, 2018-05-03:
My apologies. it is Cmd-Shift-S that is save-as, not Cmd-S