artisan-roaster-scope/artisan #196: Phases LCD defaults should be configurable or same as 1.2

peppersass, 2018-05-07:
Several issues related to the new behavior of the Phasing LCDs in 1.3.0:

1. Phases LCDs have been changed so that "time-to-DRY" and "time-to-FC" are the defaults. Although the display can be changed to the pre-1.3.0 defaults of "projected-DRY-time" and "projected-FC-time" by right-clicking the values, the default cannot be changed. Consequently, those of us who prefer the pre-1.3.0 behavior must change the Phases LCDs every roast. Please make the behavior configurable or return to the pre-1.3.0 default.

2. Prior to 1.3.0, the FC Pahses LCD switches from "projected-FC-time" to %DEV when the First Crack Start event is clicked. Now it only switches when if the operator clicks on the value in the box, which is an extra step (found this by accident.) Please make the behavior configurable or return to the pre-1.3.0 default.

3. The fly-out help for the Phases LCDs doen't match the behavior. It says to right-click to cycle between TIME, TEMP and PERCENT. It looks like the DRY% LCD switches between % and TEMP, but the label doesn't change (stays at DRY%). The other two LCDs cycle between "time until" and "time when". The fly-out help should be changed and the DRY% header should change as apropropriate.

MAKOMO, 2018-05-11:
Thanks for your suggestions and report. I just changed the behavior of the PhasesLCDs in percentage mode back to that of v1.2, rendering again "projected-time" instead of "time-to". I also fixed the rendering of the first LCDs label that is now properly indicating the mode also before TP.

I did not introduce a new setting to switch between "projected-time" vs "time-to" not to make things even more complex and decided to render "time-to" values only in temperature mode. Note that this extra information is displayed instead of an empty LCD thus there is no need to update the fly-out help as there remain just three modes, Time, Temp, Percent. Just that for example, until the DEV% ratio can be displayed as FCs is not set, this 3rd LCD is used to display the projected time of FCs.

Note that the selected PhasesLCD mode was always persisted correctly over application restarts also in v1.3.

MAKOMO, 2018-05-11:
You might want to try an actual beta build of v1.3.1 from bintray to see if this behaves now as you suggests: https://bintray.com/artisan/artisan-artifacts/artisan/master#files

peppersass, 2018-05-11:
Thanks, Marko! I downloaded and installed 1.3.1 for Windows and Mac, which installed and ran OK. Will test the Phases LCDs on my next roast.

closed 2018-05-18 by MAKOMO