artisan-roaster-scope/artisan #197: Artisan 1.3 Crash on Windows 10

arkenamun, 2018-05-08:
## Expected Behavior
Start new roast (with ctrl N, though may not be significant) with Artisan at OFF or ON status

## Actual Behavior
No loss of data, existing roast data prompted to save.
'Phidget IO 6/6/6 attached' message displayed
App. crash circa10% of times.
Otherwise makes it to ON

## Steps to Reproduce the Problem

  1. ctrl N
  2. save existing roast (alog & pdf), data saves 100%
  3. app may crash, does not make it to ON

## Specifications

  - Artisan Version: 1.3.0
  - Artisan Build (number in brackets shown in the about box):95bf230b , and previous 1st general release
  - Platform (Mac/Windows/Linux + OS version): W10pro V1709, OS build 16299.309
  - Connected devices or roasting machine: Roaster 1 (HSR)
[#HSR 1048 Production v91.9 [start stop drum available palette 2].aset.txt](https://github.com/artisan-roaster-scope/artisan/files/1983483/HSR.1048.Production.v91.9.start.stop.drum.available.palette.2.aset.txt)
[Proaster v12.4 [axes, buttons] - Copy.txt](https://github.com/artisan-roaster-scope/artisan/files/1983485/Proaster.v12.4.axes.buttons.-.Copy.txt)
[HG15-1570_18-04-30_1630.alog (background roast during crash).txt](https://github.com/artisan-roaster-scope/artisan/files/1983492/HG15-1570_18-04-30_1630.alog.background.roast.during.crash.txt)
[image]

: Phidgets 1045,1048,hub0000,out1100, Extech 755, Modbus RS485 serialport/USB adapter. Roaster 2 (Proaster): Phidgets hub0000, tmp1101
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-05-09:
This is an amazing complex hardware setup and configuration which cannot be duplicated here. Could you please come up with a minimal setup (minimal number of devices an configurations) to allow us to duplicate this issue here? Does the issue appear only on the HSR setup or also on the Proaster setup?

arkenamun, 2018-05-10:
Thank you Marko,

Good points and a good idea to minimise the configuration complexity for duplication there. Both the HSR and Proaster setups are having the same issue. The Proaster configuration is way simpler, it has a HUB0000, OUT1100 and TMP1101. I shall reduce this configuration and report back. Maybe you can also confirm there are others using those 3 Phidget devices simultaneously, if so maybe I should also try a different PC?

Just reviewing the W10 application error log, I can see the issue occurred 3 times in 11 roasts with the Proaster sample roaster setup on Monday (7th May). However, it seems I've had Artisan app crashes 25 times since 8th March. Maybe the details logged by Windows help?

[image]
[image]

22 of these errors in ntdll.dll e.g:

Faulting application name: artisan.exe, version: 0.0.0.0, time stamp: 0x5a2e9f14
Faulting module name: ntdll.dll, version: 10.0.16299.248, time stamp: 0xeffc9126
Exception code: 0xc0000374
Fault offset: 0x00000000000f87bb
Faulting process ID: 0x2a74
Faulting application start time: 0x01d3cffb952f55f1
Faulting application path: C:\Program Files\Artisan\artisan.exe
Faulting module path: C:\WINDOWS\SYSTEM32\ntdll.dll
Report ID: 7e67406a-eac9-4d23-996c-bb642550e561
Faulting package full name: 
Faulting package-relative application ID: 

3 errors in Qt5Core.dll e.g:

Faulting application name: artisan.exe, version: 0.0.0.0, time stamp: 0x5a2e9f14
Faulting module name: Qt5Core.dll, version: 5.9.3.0, time stamp: 0x5a0c7d97
Exception code: 0xc0000005
Fault offset: 0x0000000000142ac1
Faulting process ID: 0x17c8
Faulting application start time: 0x01d3bfaa35db4ce3
Faulting application path: C:\Program Files\Artisan\artisan.exe
Faulting module path: C:\Program Files\Artisan\Qt5Core.dll
Report ID: 6c19b21a-0929-4eb0-92f8-4db3c9021c27
Faulting package full name: 
Faulting package-relative application ID: 

NB, The Proaster Phidget port setup is as follows;

Phidget HUB0000
0-OUT1100
1-TMP1101 (ET,BT,AT)
2-
3-PWM Command pulsehub(3,1000); single bit memory set.
4-connected to 3; Input for alarm rule to detect 3 set
5-

Phidget OUT1100
0-
1-
2-PWM Command out(2,{}); PWM electric heater SSR
3-PWM Command out(3,100); turn on roaster power/motors (Default Buttons ON)

arkenamun, 2018-05-10:
I reduced my Proaster configuration. I seem unable to replicate the error when I only remove the device Phidget HUB0000 IO 45 from Extra Devices assignment. Without this line I tried 20 roasts without crash. I shall try more tomorrow to see if it was maybe a lucky run.

Here is the config screen before removing the device on line 1:

[image]

MAKOMO, 2018-05-11:
I am using multiple Phidgets for input and output in parallel for my setup as many other do and that works stable. However, your application seems to be rather specific. The hubpulse you are issuing has a complex implementation that involves parallel processing on several parts. 

I just tried to model the core of your issue here using my HUB0000 connecting port 3 and 4 as you do. See the attached settings. This works for me without problems on my Mac. Could you test that on your side? I guess these crashes are Windows specific as they occur in the Windows kernel and could be the result of some bad interaction of the parallel processes Qt spans. Not much to do about this on the Artisan side.

[image]

As you see in the above picture, the pulses triggered via the Pulse button are correctly recorded. I did not add an alarm rule to this, but I have no doubt that this one would trigger correctly.

[artisan-settings-pulse.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/1995433/artisan-settings-pulse.aset.zip)

MAKOMO, 2018-05-12:
This will be rather difficult to resolve as it happens obviously in the Phidgets driver interacting with the Windows kernel. Most likely this is an issue of the Phidgets driver (a new one was just released btw). I only found this related issue but addressing v21 of the Phidgets driver which is based on a totally different code: https://www.phidgets.com/phorum/viewtopic.php?t=7705

arkenamun, 2018-06-20:
1) Testing your .aset configuration file on Windows 10

I tested the configuration file you supplied, it runs without crashing on Windows. I tried 40 Starts.
Adding another Phidgets device to extra devices in this configuration: I had a 1048 available and added 1048 23 device. No other changes. Crashes then happen regularly again.

[artisan-settings-1.aset.TXT](https://github.com/artisan-roaster-scope/artisan/files/2117912/artisan-settings-1.aset.TXT)

The crash always happens at the same point "6/6/6 attached" displayed. I suspect the HUB0000 device, in conjunction with a further Phidgets device is required to cause the problem

2) Issue appears RESOLVED by WORK-AROUND in production (with complex device configuration, W10)

I have undertaken 34 roasts since removing HUB0000 in production, without crash.
HUB0000 and OUT1100 are replaced with 1018. Initially I had the maximum 10 (x2) extra devices including 3 Phidgets temp devices (1045IR,1048,1051) plus IO67 and IO Digital 67, plus modbus and virtual without issue. Now a slightly reduced configuration is running more smoothly (initially rules at ON may not fire until START, but now working reliably).

I note, that unlike 6/6/6, there is no visible message for 8/8/8 displayed. And unlike when using 6/6/6 there are no crashes, even with a more complex configuration, and more physical Phidgets USB/Artisan devices.

Phidget22-x64_1.0.0.20180522.exe
1.3.1 (3001eb91)

MAKOMO, 2018-07-07:
Sorry for the delay. I just tested my artisan-settings-pulse.aset again adding a second extra device of type 1048 (channel 01 or 23) and this works for me (under Mac OS X) without any crash (using the latest Phidgets driver). So I guess there is something wrong on the Phidgets driver side for Windows for this configuration. Seems to be nothing that can be fixed on the Artisan side. Sorry.

MAKOMO, 2018-07-10:
I should have done before: I now tried the test with the artisan-settings-pulse.aset with one extra device of type 1048 channel 23 added on a Windows VM (I don't have access to a PC) running Windows 8. So the VINT HUB with chan 3 connected to 4 and an 1048 are connected via USB. No crash running Artisan v1.4 from current trunk under Windows. I am running out of ideas on how to re-produce this. The Phidgets driver installed on this machine is rather old: 1.0.0.20171120 (64bit).

MAKOMO, 2018-07-21:
We decided to close this for now until we found a good and simple reproducible test that fails

closed 2018-07-21 by MAKOMO